trezor/trezor-firmware #1764: Cardano Sign Transaction - button request other

vladimirvolek, 2021-08-17:
**Describe the bug**
All button requests are `ButtonRequest_Other` In Cardano sing transaction flow.
In the same flow in Bitcoin it's `ButtonRequest_SignTx` and `ButtonRequest_ConfirmOutput`

**Firmware version and revision**
2.4.1

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Suite
 - OS: macOS
 - Browser Chrome
 - Version 92.0.4515.131

**Expected behavior**
No `ButtonRequest_Other` in the sign flow.

tsusanka, 2021-08-18:
We want to rework the button requests altogether. How critical is it for you? Also @gabrielKerekes would this be okay on your side?

matejcik, 2021-08-18:
related: #35

gabrielKerekes, 2021-08-18:
> Also @gabrielKerekes would this be okay on your side?

To be honest, I've never had to deal with the Button requests other than in the tests at some point - but we've already removed the expected responses. I think changing it shouldn't cause any problems on our end.

tsusanka, 2021-08-19:
@vladimirvolek do you think you could open a PR for this? It should be straightforward, just search for `.Other` inside the cardano app. Example here: https://github.com/trezor/trezor-firmware/blob/395324a8ad9399bacba2ebb8740d72971842d761/core/src/apps/cardano/layout.py#L85

It would be better if you do it, so it fits your needs.

tsusanka, 2021-10-07:
Let's solve this via #35 (the new button requests).

komret, 2023-05-30:
Seems like #35 is not moving forward and this bug is forcing some [weird workarounds](https://github.com/trezor/trezor-suite/pull/7964/commits/f243277668880fa4e58fb206a05f1ec9e075c157) in Suite. Can someone look into it, @Hannsek?

Hannsek, 2023-05-30:
Does Suite need this right now or it can wait few weeks?

komret, 2023-05-30:
Not urgently, we have the workaround, it's just ugly.

ibz, 2025-04-24:
Reopened due to #18501 which happened as a follow-up of https://github.com/trezor/trezor-firmware/commit/b8e2709ca8f141a5ded1ffdd5d20d4b388fe49d6