trezor/trezor-firmware #1767: Re-enable Firo support. Solved scriptSig size limits

reubenyap, 2021-08-19:
Firo (formerly Zcoin) support was dropped because Lelantus spends to the Trezor made it unspendable. This was due to Lelantus using scriptSig and exceeding the size limits. See issue #1588  

We have rectified this issue and implemented the recommended fix which is to move this data out 
https://github.com/firoorg/firo/pull/1063
https://github.com/firoorg/firo/commit/2cfa8e9123a8be462a41e60f70039d721c863f38

This requires a hard fork which will be completed by block 401580 (approximately August 26 2021).
https://github.com/firoorg/firo/releases/tag/v0.14.8.0

The only thing that needs to be done is to ensure the firmware has support for extra payload for Firo which I believe is already enabled. As long as it's enabled, nothing else needs to be done.

@matejcik

matejcik, 2021-08-19:
Is there a running testnet where we could verify that this works?
Can you prepare a prevtx that looks like [this one](https://explorer.firo.org/tx/bc9f34adcd45944fc1cb3c6b806f3cacd291e1ef75441691a59758fd368f75d1) but with the new way of encoding shielded outputs -- and ideally send some coins to an address generated from [`all all all...` seed](https://github.com/satoshilabs/slips/blob/master/slip-0014.md)?

matejcik, 2021-08-19:
hmm, this still means that _existing_ shielded txes will be unspendable because they are already encoded with the original method, correct?

reubenyap, 2021-08-19:
> hmm, this still means that _existing_ shielded txes will be unspendable because they are already encoded with the original method, correct?

Existing funds that have already been sent into the Trezor using Lelantusjsplit transactions as input would be unspendable and would need to be recovered using the seed. Don't think there can be any workaround for this but I believe this should be a minority of users from the support requests we have received compared to those locked out from losing firmware support.

However, any new funds coming from Lelantusjsplit should now work once the hard fork date rolls around.

Our existing testnet is already setup to test this..

Please view the transaction on testnet [here](https://testblockbook.firo.org/tx/c001dc5aacf3da41114c293eec22bb671fb9cb73566bd006b7c98ca07e5ec619).

reubenyap, 2021-08-19:
```
{
  "hex": "03000800010000000000000000000000000000000000000000000000000000000000000000ffffffff01c9ffffffff020001b2c4000000001976a914a1951d50c386c6e328908d2f6a1ff9ddaf7305b388ac000000000000000053c696ed949a0183285f847d4b521205d0502d6d9ccee5baf7f3ea285f16a9a8ddfa0100415540659581d24773e0e812c1e5e8bfa32e6f4750105494817167c85b3eba8b7494bf87afcf30fe80c49f84a06a0f6e108f0000fd1d0e01e0b31c7f93dfb84b223d3946653b237c911da3a63a0d43aff16440b2677edf1b00009e13bcf58dd14e51c65e92f5568111604baf28d5615604923c44f85cc9cdb41d00000293005c6b3b6521118d5203e634ffd8dd4b9bf37d47e4b8f70f45a10cfd641800000163e6502f9f812c86d572e5ec0d29879ac631f676dbb6f06af105925bb23b8300003c8c494575f6aab997dd6652de7ff94c7cdfd9ee44fd208068dc0739a6c6bc3273d2d293c6c650ed7bbba8e248d275bd9262858902e8682e9f7605232d7aed45f48f9be2b243967343caac9cc29b78e481740bef0faa153b24857aaeeae9b1fe635e363f319590e638e9e391b713fc5bb87c7d7f051667bf905517e6a454aa4b1cd1b7f0c6597b59f9d7352c53ac9a7195d8d4f8814bd137920b85841512871e75b20d630c111672e942539282e9b03022dacf7aea11f0117a58b0ea8bdfbcbf2b9c2bd1e70d4d89a86609b762d3e1fd836e5b56329a187fc57e42004a33104ff2215db407c730c63755a6f2359283aece5917ea9307d7b89756c74a9e43154f9e885d2d8287fe80770199efd225e0801198ea35a8bb27eeb2fc9b39aeb514ac340c22188451b29d65244857b9e341200a2b233a304b6ecf1e563be03085cdc06afff4ef4bc82d0225520bf5d18a8bd3e62abe9d87699f63c2117bc69e8a3053d8a74fac472618e608f00374866bab43419fbeccfab010392b4e8109a53fd56586db27ec1cbd02bb564a9640f02d3ab2ba3f65b828f55bd36234b33de41ee58d0409aa9b01b464a219f0dc885df1d24607f9c5251e74e3722423223f34c59a27f55cc439dafdb53bddf1951fd7092bad01c285de57b5e91a5b48340524cfdb91a4e84afb9c9caba683bca946afda3593dc94d38bac156c699642ce98fa7c212211ddbf4abd2c72bafbe41b6c5bfa419d4cef534460be82b979503a6a8c0cc428ffd0c9c276a151e6b61056c98698d00e56465d00daea0a70ac8df40fb41b9d716f14c8ca77299fae7ba94180adf819b3e1bebb4afd96e13774170ef85d640795aec3c257e96ba7e2407085001b1345a1c4b7bea9258fad12a0cfd2f1bc7b0d9784459545dcc4035af7ff34ad4ed0758a69c6180c57a252c331fd71eee4e3d5ad509fe5920887b0b9126482c70e47ac19eec3c3dbdb5d78d458018b3058df4d7bb1c75c2d3ae7681eb3155b528092ff57b21c24bcb8f1f1bfb83ca50c22e3018eb1a8235db1add97a3539296ef1f7def6b4e0d3648690b4f13fca7280fc287c0fb81f85ee72a26dbe97aebecb5095d52f1f6ae2525cbfa66b7c90d2d008f0289ca4716dc75a5c7dba06c9a36964229cc7de37c2cb888ac0709a92d64399ce12110daa3006aeea94164d9356b4f55d744eb53a82f6d9f8a79e8f006183afc375756ea868f7d52f665ec8c64d720360c5faae00ac429edf2eccef735f1925cdbb369e90b9f6a72f67610650388db047bc7adb213a7da93a3de0b7d22c4f5c7f13fd559a3b97c3ae094408e49ef3942a9719e51864bbfcc68a9c7d9db7c0d4f328c0728108aa6731d2c54a8baf26e376c45834211c3d963558d61f027dde841fcff67c6adf359a25f76eb5abd41ed485f41047c6770dc52ffc2670f64ef32475723f8c2fe4bd8098341569e41be2b8c9c2d99e33a0d7d161168c9842245e988ed1519d4884079e6bae48232afb1d156aef560e6ad3a75066b9593974da6f4dea21dbb2d92a47bc95eb5a10c097b0f87cb82c6e68253828eee64e289f922a097a3f1cfe43568d8d2c60bb5a146d06fcc8c2d6d907ce433196630adaec63eff73cf8eee329c2bb92906d5e29545880158232b326970b07918f12dd4bbff8b8074915a24dd1d23b6a47e8df493f6c6e25e0a6bc02b6fc6ca2fdb904451bb86347e48fa31b48112a6fe37dda4e7a041737d1ae52a7764794a6dcbdda9b9c662b6774b896d6818456bd133eebf0d7780ba455383c08e0ff0f5c47ad5fd6ee765ca77ddc5d0d1da4d3a624e01d587093a07ba11b0f0e82eaf76747694c5db6f5d8fd6bbe7bb05af6c7ad97ef8d39e0f29653876bd1fdda325647b12cbc3c74942964a817b0777058eae01cc4161a50d4b512561b95f3bff2bb1fc8fc383aeebd634f38597cfcbe266b900a3a085a016f7fb85bc781daff630d79d015845b629ac28a5735b851f39381af78936bb7bce39cc9b70c50c499a47cfa10a150363b0088cf4b5944528e2a50e961728373f05546182f8599c6134d0486beaae1d1147141032e66f1491f760048b3547954e1afa5a0af9fed1bda2a5a9a64be6e11318e35c08843bfc8929ef560a2e5cbcc2fb5ab56d121b416ebbafe12ef9f846d888a7fae9a43916dcfa7d427ea26d338154652fc6b8ef88b248a97a154a200c634e83d3237de7df2bbaf0023736398232443ebc6fa64407ef3f5c45a495d0662c9a8147a2590d4c3a786e2b20107299eea70ff4068a925efe02c2952e23d7b91e16e80b1b5c37e7f0ff53b9e62604c3a1a2de8b90972b8d3d66dfb6cd03d23b24b9802be696b452d15198156ced0e8f3ebc6cae3351d7be65e4eae42adeff127499b3efd500c16e845b4652c4949ad95c22bfaf4fd2cb7240c17ba77d70a70062a0ac17af824fab0f935759528d73a02c6cb428d836291c00cb1aa4a5aa8ea7914327082cc9ba0502e0e653b802cb4c12b4829218f410bbca5e68b368bc6feeeaaa0f4fddfaa9102fb8ee80416eb89aa704b287b1c0d3dd1fbb83262da405befb033fec6d2cf50da78da38cb681a9af65dfb1770c9d829a94c1a2bb516e538a0df744c82e1a17afc4f8d6b1e69cb920ae6be7b6f45be0d99f2dc0ae22e5aef5b5f2b34b8e8165415f1628eb1d8e8ee3f629296a5d38e95f6ee6399914ddb2553dcbc46d6c0c2e23f7f53c24acde4b2dcd297f364f3e3699ec2e898147d9ffcd2f047709a3ac47be1c20338a19a969662d1d80c117e7116431d3b20728599f463133fa01f7e1c3510041332728ead994ced0a117839001213887708e6748aaca803bb85279dd710ade300005ed8383e58f20b9e5a833633bf8e9b8ff17928f5e3b676c20c472ac080e822590000d752aed302f0b70bf0b448dfcd1e709c58861ee94e0e19be55d2ce200d4522600000c57a9568ed475f0901f23ef6f7d4567f23e514968603c3fc7f9b5a5a6ac5cda2010004b7d0f23ed01e47a69731a5d3b850810d8e0a2f963c50fae635d4ae67e03a36500100591bbf4b5f32620e2b6b93f368d4c35f3ed377ec478d9bed2ce3f4ec8f74e96b010087143c8e576835fba115b5d282de9b16f2ad9830d96e264db89852e6f616717d0000b7054fca543164849ccfee97fb3c982f671443b1d7becb58428f83371f9590f5000093a04fc5e895b395b37eeaec055a60b2b85c2be5cd238d0aba57c48be6fa7d20cb023dcf844c637a9691a8337a4a188fa05d780d3f0d2862b9d6922c238f9167b1811c1bb3a31f61d31467497991ff2eb51753612590f1aa95fb27adf274782600006d28f2c221e3e8352eceadfbe1defae8e5a063c83ee49cba22262a19f36bec7600000b2e76a674acfb3d2d9784a5a7657c103e23808ff61f72bc61a43387d329fe3f00 [... cut ...] 00000bb5a7ce38c1505b312592e9a2aa6a23bec7ddadc6cf519091f1ae7c311c9793b0cefdf51d8cf88341b339ca275f9dbf66076b5d673772962cb93ec253e7e864503e5fd6c1c0e1d13c509811ebb1f9750617cd31a900ec64ebd5bc5f382f0dd028b01010000009e8cba2497140952be1b8d8749e20ccdb1b6988e8ee7013723096a01594f724268280100000000002f000000988a19f55cc81cc927a84d3cac12106e08dd66b467f0b6d6baf0b385bda25c830100dfd99b5279961bda80bec9bbec030bd6d7b5cd1186386a497fe8cdaf440cbee438c61bff758c46cc7b987d18dd7892073ae6e8c51acf88c07538eccb977a97c8"
}
```

justanwar, 2021-08-19:
Here are the two Lelantusjsplit (new format) to addresses generated from `all all all ...` seed:

[https://testblockbook.firo.org/tx/8a34ccceaf138fd14398303340afb37871cb0ea6719ceba315172edb9ff6d625](https://testblockbook.firo.org/tx/8a34ccceaf138fd14398303340afb37871cb0ea6719ceba315172edb9ff6d625)

[https://testblockbook.firo.org/tx/fdb1a55c8aaa9b3f56efa0852090272a8717544ec9303876fcc097a5ec9c6c6a](https://testblockbook.firo.org/tx/fdb1a55c8aaa9b3f56efa0852090272a8717544ec9303876fcc097a5ec9c6c6a)

matejcik, 2021-08-20:
@justanwar what derivation paths are these? I can't find them in the usual places (that is `m/44'/1'/0'/0/` and a bunch of variations on that)

justanwar, 2021-08-21:
@matejcik The derivation paths for the addresses:
```
m/44'/1'/0'/0/4 THgGLVqfzJcaxRVPWE5fd8YJ1GpVePq2Uk
m/44'/1'/0'/0/5 TPypFKi3aziXmiH1MiwagaK71apv5XARGY
```

matejcik, 2021-08-23:
thanks for your cooperation folks

commit f3a49a67 (2021-08-20): feat(all): re-enable Firo (fixes #1767)
commit be1f9454 (2021-08-23): feat(all): re-enable Firo (fixes #1767)
commit 7aed8d53 (2021-08-23): feat(all): re-enable Firo (fixes #1767)
closed 2021-08-23 by matejcik