trezor/trezor-firmware #1769: Mac and Linux generate different UI fixtures

matejcik, 2021-08-19:
as reported in #1683 and #1753 

Per @prusnak's investigation, different PNG encoding library is used, presumably with different compression level.

Options:
1. figure out the compression levels and try to force the same one
2. explicitly use libpng with same parameters on both platforms
3. instead of raw PNG data, calculate hashes of decoded image data

option (3) seems most straightforward

prusnak, 2021-08-19:
> 3\. instead of raw PNG data, calculate hashes of decoded image data

+1 for trying option 3; options 1 and 2 are far from being straightforward

matejcik, 2021-09-01:
@grdddj to implement option 3, you would need to look into `tests/ui_tests/__init__.py` function `_hash_files`, modify it to decode the image files and hash the decoded contents

closed 2021-09-14 by matejcik