trezor/trezor-firmware #1774: Allow disabling pin scrambling on Trezor Model T

ntninja, 2021-08-20:
**Is your feature request related to a problem? Please describe.**
I'd very much like to use the TT's pin lock feature, but I'm extremely bad at remembering numbers (like, really really bad – call it a disability if you must). For things like my bank card I compensate for this by remembering the pattern of buttons I have to press instead and it served me well so far.

So when buying the TT a few months ago I expected to do the same, but found that is approach is completely infeasible with the scrambled pin pad it uses and I simply couldn't cope with the scrambled input pad to the point where I had to disable the pin entry entirely and only use the passphrase feature.

**Describe the solution you'd like**
I want this line to be optional/configurable: https://github.com/trezor/trezor-firmware/blob/f37ca13f1ae952f42da8d7bfb65baf427cd4175e/core/src/trezor/ui/components/tt/pin.py#L22 (even if only through `trezorctl`)

Since the Model T is not dependant on the host device for input, I believe the security impact of adding an option for this for this device would quite acceptable – even it it means leaving potentially useful-to-attackers touchscreen fingerprints being left behind.

**Describe alternatives you've considered**
Doing nothing and leave pin entry disabled entirely on my device. :unamused:

matejcik, 2021-09-07:
this would also be useful for vision-impaired users

stepmes, 2021-12-12:
Thanks very much for this input, we're noting it down and will consider it when working on future improvements.

closed 2021-12-12 by stepmes