trezor/trezor-firmware #1775: Would you accept a proper implementation of Solana

t-nelson, 2021-08-21:
Hi!  Sorry if I'm being dense, but from the [contributing guide](https://wiki.trezor.io/Developers_guide:Contributing) it is unclear to me whether a proper implementation of Trezor firmware support for Solana would be accepted at this time.  It looks like there have been a couple community false starts that have gone nowhere.  I'm an engineer with Solana Labs and a primary maintainer of our Ledger app.  We're interested in a Trezor implementation, but would like to confirm that it's something that you have capacity for before dedicating resources.  Please advise.

Thanks!

mdhz, 2021-08-21:
Very good opportunity here for the Trezor team to work with. It's been the most requested chain from your users. Please do something. It's top 10 in the total market cap now.

zhangcanaaa, 2021-08-22:
Totally. If I had to put my hands up, I'd put my hands up and my feet up!

karamellar, 2021-08-22:
It's a shame that sol is missing considering it's such a popular chain

xiaoyu2006, 2021-11-02:
Any updates?

Solana had grown quickly in the past week, so this altcoin should be a little more priority IMO.

penilessquant, 2021-11-15:
2nd that. I love my Trezors but I am doing more and more in the SOL space and unfortunately am moving to the dark side...

kdckrs, 2021-11-17:
@tsusanka Any news about this? SOL is now the 5th biggest token (market-cap wise) so it should be a big priority to add compatibility.

g00nix, 2022-01-01:
I was about to buy a Trezor but I see it has no support for Solana. As soon as you add support, I am buying.

abh1, 2022-02-26:
Same, need to secure my sol...trezor wont do

alifarooq0, 2022-10-24:
Hello @prusnak, @tsusanka -- does the Trezor team have plans to support Solana in the upcoming Trezor model?

amplicity, 2022-12-09:
Has there been any progress here?

Hannsek, 2022-12-09:
In near future, unfortunately, Trezor will not support Solana.

amplicity, 2022-12-13:
@Hannsek thanks for clarification --- any reasons as to why not, or is it just not a priority at this time?

Hannsek, 2022-12-13:
No particular reason. Not a priority right now.

cac-99, 2023-11-02:
Adapt or die, it's the free market.

matejcik, 2023-11-02:
closing in favor of #3359

closed 2023-11-02 by matejcik