trezor/trezor-firmware #1778: Improve backup verification

Giszmo, 2021-08-24:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when I have to type in 24 words just to know which of the two backups I hold in hand is the right one.

**Describe the solution you'd like**
Let me verify parts of backups, only. First word for example.

I understand the Trezor might only stores the derived masterseed and not the actual "words", so this might not be possible without *also* storing the words. Either way I'm curious if this could come with an update.

matejcik, 2021-08-24:
The purpose of "dry-run recovery" is to check that you have backed up your seed correctly, which is only possible if you enter the full backup.

If your usecase is "matching correct backup to correct Trezor", I suggest using some other method, e.g., labeling your backup sheets.

closed 2021-08-24 by matejcik

Giszmo, 2021-08-25:
> If your usecase is "matching correct backup to correct Trezor", I suggest using some other method, e.g., labeling your backup sheets.

Your "solution" cannot be applied once the trezors are mixed up. Pretty arrogant to suggest I had better not let the friends mix their devices and backups. They also had to rotate backups occasionally, so there is even more backups floating around, where they are not sure which is which. :man_facepalming:

prusnak, 2021-08-25:
The solution is to enter the whole seed. That's totally acceptable solution for someone who misplaced their seeds.

matejcik, 2021-08-25:
@Giszmo again, the dry-run recovery feature is designed to do something other than what you are trying to do.
There are security implications, too: for starters, allowing checking any N starting words of the seed would allow someone to extract the whole seed in 24*1024 attempts, or maybe even only 24+23+22+... attempts if they previously captured the scrambled words from Standard recovery.

We _could_ of course design a feature that only allows comparing the first word of the seed, for the express purpose of matching backup to device. But then you have 1 in 1024 chance of a collision, and there are several other options that work just as well and do not touch the seed, a.k.a., **the most security-critical part** of Trezor.

One option is to set an unique label on Trezor screen, or a unique homescreen. (If you're OK with reducing security by one word, you could set the first word of backup as your label.)
Another is using the device ID, which can be shown in Electrum or in the url at wallet.trezor.io.
Yet another is using distinct silicone cases for the Trezors, or putting stickers on them.