trezor/trezor-firmware #1780: Duplicates in coins_details.json

prusnak, 2021-08-24:
There are duplicate entries in coins_details.json:

``` json
        "eth:ONE:1666600000": {
...
        },
        "eth:ONE:1666600001": {
...
        },
        "eth:ONE:1666600003": {
...
        },
```

``` json
        "eth:QKC:100000": {
...
        },
        "eth:QKC:100001": {
...
        },
        "eth:QKC:100002": {
...
        },
        "eth:QKC:100003": {
...
        },
        "eth:QKC:100004": {
...
        },
        "eth:QKC:100005": {
...
        },
        "eth:QKC:100006": {
...
        },
        "eth:QKC:100007": {
...
        },
        "eth:QKC:100008": {
...
        },
```

We already know about these, because they are already listed in [`common/defs/duplicity_overrides.json`](https://github.com/trezor/trezor-firmware/blob/master/common/defs/duplicity_overrides.json).

matejcik, 2021-08-24:
we will need to modify the format of `duplicity_overrides` so that it tells us which coins belong together

prusnak, 2021-08-24:
> we will need to modify the format of duplicity_overrides so that it tells us which coins belong together

How about switching `true` to `null` and `false` to string?

```
{
    "erc20:eth:BTL (Battle)": null,
    "erc20:eth:BTL (Bitlle)": null,
    "eth:ONE:1666600000": "eth:ONE:whatever",
    "eth:ONE:1666600001": "eth:ONE:whatever",
    "eth:ONE:1666600002": "eth:ONE:whatever",
    "eth:ONE:1666600003": "eth:ONE:whatever",
}
```

matejcik, 2022-12-13:
superseded by #2410

closed 2022-12-13 by matejcik