trezor/trezor-firmware #1782: ETH ERC20 tokens not indentified

bosomt, 2021-08-25:
**Describe the bug**
ERC20 contracts/tokens are not identified when sending transaction with latest freeze version 2.4.2
Works as expected with latest public version of 2.4.1.
TUSD token is identified.

https://eth1.trezor.io/address/0x0000000000085d4780B73119b644AE5ecd22b376

**Firmware version and revision**
2.4.2 revision 0xdd6578061035bdaea2932fda363abde6198640ef

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Suite 21.8.1
 - OS:macOS

**To Reproduce**
Steps to reproduce the behavior:
1. Try to send ERC20 transaction with 2.4.2 firmware version
2. Try to send ERC20 transaction with 2.4.1 firmware version
3. Compare device behaviour.

**Screenshots**

[image]

[image]

[image]

matejcik, 2021-08-25:
i believe this is TUSD-specific, please try other tokens too

matejcik, 2021-08-25:
What happened: our duplicate symbol detection kicked out TUSD because there is a `tusd` Ethereum network in the `networks` repo. That is however a testnet which is auto-unsupported, so this shouldn't have happened.

In any case there should be a cross-check that we are not kicking out, say, the top 100 tokens?

bosomt, 2021-08-25:
@matejcik  what a coincidence ;) 
yes it's TUSD related 💯 
i was not able to reproduce issue with different tokens !