trezor/trezor-firmware #1783: Fix Cardano compatibility issues

andrewkozlik, 2021-08-25:
There are a number of different seed derivation schemes used in Cardano, i.e. the way in which the mnemonic is converted into a seed. The officially recommended derivation scheme is Icarus, see [1]. In Trezor we implement Icarus, but the implementation has a bug which manifests in case of a 24 word seed, see https://github.com/trezor/trezor-firmware/issues/1387#issuecomment-798790044. So someone who is trying to import or export a 24-word Icarus seed between Trezor and another Cardano wallet will fail [2]. Ledger has their own derivation scheme, which is BIP-39 based, and incompatible with everything else [3].

It would be helpful to users if Trezor supported all three schemes and in Suite users would be able to see all their accounts, similarly like we do with legacy, segwit and native segwit accounts in Bitcoin. The default account type is native segwit, but if a user has coins on a legacy account then they can still see them and spend them or create a legacy account if they wish. Similarly, in Cardano the native account type should come from Icarus derivation, but Suite would automatically show the other two account types if the user has previous transactions on them.

The way to implement this would be to add a field called `CardanoDerivationType` to each Cardano protobuf message that has an `address_n` field. The derivation type would dictate how to obtain the seed.
```
enum CardanoAddressType {
    ICARUS = 0;  // The future default.
    ICARUS_LEGACY = 1;  // The current implementation with bug.
    LEDGER = 2;
}
```

This issue is related to https://github.com/trezor/trezor-firmware/issues/1231 but is orthogonal. They can be implemented independently.

[1] https://github.com/cardano-foundation/CIPs/blob/master/CIP-0003/CIP-0003.md
[2] https://forum.cardano.org/t/all-ada-cardano-seem-gone-account-shows-0-with-trezor-model-t-ledger-nano-s/35784
[3] https://forum.cardano.org/t/swapping-from-ledger-to-trezor/50513/7

tsusanka, 2021-08-25:
Btw `LEDGER` way would basically solve https://github.com/trezor/trezor-firmware/issues/1231, right?

andrewkozlik, 2021-08-25:
> Btw `LEDGER` way would basically solve #1231, right?

Yes, but users would need to migrate and you'd have to have some intermediary firmware which would support both. Doesn't seem user-friendly at all. Might as well support both the whole time. Also you don't want to force user's into a vendor lock-in, r in this case a hardware wallet lock-in, so you should ideally support Icarus as default.

matejcik, 2021-08-25:
Supporting both Icarus and Icarus-Trezor simultaneously would add yet another 96-byte field to the cache. This would grow the total cache size from 2770 bytes to 4190. We currently have 192 kB RAM available, so the cache would take up 2 % of that. With the memory savings currently in the pipeline, it seems we can afford that.

---

#### Details and some further thoughts

Items currently in the cache are:
* 32 bytes for session id
* 64 bytes for master seed
* 50 bytes for passphrase for Cardano
* 1 byte for Monero live-refresh flag
* 128+2 bytes reserved for preauthorization data

for a total 277 bytes per entry, so 2770 bytes with 10 cache entries.

Solving #1231 the way we currently intend changes the 50-byte passphrase entry into a 96-byte Cardano root, moving the total to 323 per entry.

Implementing this would add another 96 bytes for a total of 419, or 4190 for the whole cache.

Further optimizations available to us:
- reduce session id to 16 or maybe even 8 bytes
- move preauthorization data to a separate cache
- reduce number of sessions

It might also be possible to allocate (and reallocate) session entries on demand: we want all cache entries to live at start of memory, so that they don't cause fragmentation. However, cache entries are created at well defined places, and we (try to) clear memory as best as we can after every workflow. It might be possible to insert items into cache and then tell the cache to reallocate the new data when the workflow ends, thus placing it at start of memory.

matejcik, 2021-08-25:
alternative: instead of specifying derivation per Cardano call, specify it per session:
```
message Initialize {
	optional CardanoDerivation = x [default=LEDGER];
}
```
In Suite settings, you can switch the kind of derivation you want to use for Cardano, plus there are ideally some easy migration buttons.
Pro: cleaner of FW side
Con: less discoverable for users, more work for Suite

tsusanka, 2021-08-31:
So trying to make a summary from our product and standup discussions:

- We will send a _hint_ in the Initialize message ([comment above](https://github.com/trezor/trezor-firmware/issues/1783#issuecomment-905565535)) noting if the Icarus Cardano derivation should happen [1]. This will depend on the fact whether Cardano is enabled in Suite, or not.
- We consider Icarus ("non-Trezor") the default derivation.
- Suite will make a discovery for all derivation types (Icarus, Icarus-Trezor, Ledger). If Icarus-Trezor and Ledger are empty they are not shown, only Icarus accounts are displayed.
- In all Cardano messages, we need to send the derivation type as well. If a mismatch between what was in Initialize and what is being sent now occurs, firmware will throw an error. This should not happen in Suite, however.

[1] It is up for discussion whether this should be a boolean, or a list.

prusnak, 2021-08-31:
> * Suite will make a discovery for all derivation types (Icarus, Icarus-Trezor, Ledger).

That also happens if and only if Cardano is enabled in the Suite, but I guess that is obvious. (Same as the first point).

matejcik, 2021-08-31:
Suite must properly check that the discovered accounts are different. This btw leaks information about (1) if the user has SLIP39 backup (which is shown in features anyway) and (2) if the user has 24-word seed (this is not shown :) )

With SLIP39 backup the derivation-type parameter is ignored.

Derivation type should be specified per `CardanoSignTx`, not per `address_n` -- this simplifies implementation, and cross-derivation spending is a weird usecase anyway. I vaguely remember that the current signing code doesn't even allow cross-account spending.

There should be no protobuf-level default, the derivation type is a `required` field.
This breaks 3rd parties, but they would already be broken because they don't specify the right `Initialize` flag.

=> we should roll this out alongside #1231, so we only break it once :)

We probably don't need to do anything in UI, and can just show the derivation path as before?

closed 2021-11-10 by matejcik

matejcik, 2021-11-10:
QA: this will only work after the matching Connect PR is deployed: https://github.com/trezor/connect/pull/933
Check that in Adalite and Yoroi (do we know any other Cardano wallets with Trezor support), the receive address is identical before and after FW upgrade, with:
* 12-word seed
* 24-word seed
* Shamir seed

STew790, 2021-12-08:
**QA OK**

Just tested this with @bosomt

lihaibh, 2022-01-14:
@matejcik 
Ive created a 24 word seed on trezor connected it to yoroi and then tried to restore the same wallet using the same seed and received different wallets.
Im not sure this issue resolved. Im using the latest firmware version: `2.4.3`

PR #1873 (2021-11-10): Cardano derivation improvements
With this PR, the same passphrase is used for Cardano derivation and the standard derivation (fixes #1231). If enabled via `Initialize(derive_cardano=True)`, the Cardano derivation runs at the same time as the standard derivation, which means that the start-up time is longer.

This is only relevant for BIP-39 seeds. With SLIP-39, a different derivation is used and it is available even without `derive_cardano=True`.

Additionally, the Ledger and Icarus derivations are implemented. Ledger derivation is available even in sessions that don't have `derive_cardano=Tru [... cut]