trezor/trezor-firmware #1787: mimatched bound error when compiling trezor core on linux

achow101, 2021-08-28:
When compiling trezor core for linux (with `make build_unix`), I get the following build errors:

```
vendor/trezor-crypto/sha2.c:580:46: error: argument 2 of type ‘sha2_byte[]’ {aka ‘unsigned char[]’} with mismatched bound [-Werror=array-parameter=]
  580 | void sha1_Final(SHA1_CTX* context, sha2_byte digest[]) {
      |                                    ~~~~~~~~~~^~~~~~~~
In file included from vendor/trezor-crypto/sha2.c:33:
vendor/trezor-crypto/sha2.h:95:28: note: previously declared as ‘uint8_t[20]’ {aka ‘unsigned char[20]’}
   95 | void sha1_Final(SHA1_CTX*, uint8_t[SHA1_DIGEST_LENGTH]);
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~
vendor/trezor-crypto/sha2.c:634:40: error: argument 2 of type ‘char[]’ with mismatched bound [-Werror=array-parameter=]
  634 | char *sha1_End(SHA1_CTX* context, char buffer[]) {
      |                                   ~~~~~^~~~~~~~
In file included from vendor/trezor-crypto/sha2.c:33:
vendor/trezor-crypto/sha2.h:96:27: note: previously declared as ‘char[41]’
   96 | char* sha1_End(SHA1_CTX*, char[SHA1_DIGEST_STRING_LENGTH]);
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gcc -o build/unix/vendor/micropython/py/objstr.o -c -DSCM_REVISION='"\xeb\x34\xc0\x85\x0e\x8b\xc7\x48\x52\xb5\xf8\xac\xa5\xc3\xab\x78\xdc\x86\x37\x96"' -DPYOPT=1 -DBITCOIN_ONLY=0 -DSTATIC= -Os -g3 -std=gnu99 -Wall -Werror -Wuninitialized -Wno-missing-braces -fdata-sections -ffunction-sections -Wno-sequence-point -pthread -DTREZOR_EMULATOR -DTREZOR_MODEL=T -DMP_CONFIGFILE=\"embed/unix/mpconfigport.h\" -DAES_128 -DAES_192 -DUSE_BIP32_CACHE=0 -DUSE_KECCAK=1 -DUSE_ETHEREUM=1 -DUSE_MONERO=1 -DUSE_CARDANO=1 -DUSE_NEM=1 -DUSE_EOS=1 -DTREZOR_FONT_BOLD_ENABLE -DTREZOR_FONT_NORMAL_ENABLE -DTREZOR_FONT_MONO_ENABLE -D_REENTRANT -Ibuild/unix -I. -Ibuild/unix/embed/rust -Iembed/rust -Ibuild/unix/embed/unix -Iembed/unix -Ibuild/unix/embed/extmod/modtrezorui -Iembed/extmod/modtrezorui -Ibuild/unix/vendor/micropython -Ivendor/micropython -Ibuild/unix/vendor/micropython/ports/unix -Ivendor/micropython/ports/unix -Ibuild/unix/vendor/micropython/lib/mp-readline -Ivendor/micropython/lib/mp-readline -Ibuild/unix/embed/extmod/modtrezorconfig -Iembed/extmod/modtrezorconfig -Ibuild/unix/vendor/trezor-storage -Ivendor/trezor-storage -Ibuild/unix/vendor/trezor-crypto -Ivendor/trezor-crypto -Ibuild/unix/vendor/micropython/extmod/uzlib -Ivendor/micropython/extmod/uzlib -I/usr/include/SDL2 vendor/micropython/py/objstr.c
vendor/trezor-crypto/sha2.c:887:50: error: argument 2 of type ‘sha2_byte[]’ {aka ‘unsigned char[]’} with mismatched bound [-Werror=array-parameter=]
  887 | void sha256_Final(SHA256_CTX* context, sha2_byte digest[]) {
      |                                        ~~~~~~~~~~^~~~~~~~
In file included from vendor/trezor-crypto/sha2.c:33:
vendor/trezor-crypto/sha2.h:103:32: note: previously declared as ‘uint8_t[32]’ {aka ‘unsigned char[32]’}
  103 | void sha256_Final(SHA256_CTX*, uint8_t[SHA256_DIGEST_LENGTH]);
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
vendor/trezor-crypto/sha2.c:941:44: error: argument 2 of type ‘char[]’ with mismatched bound [-Werror=array-parameter=]
  941 | char *sha256_End(SHA256_CTX* context, char buffer[]) {
      |                                       ~~~~~^~~~~~~~
In file included from vendor/trezor-crypto/sha2.c:33:
vendor/trezor-crypto/sha2.h:104:31: note: previously declared as ‘char[65]’
  104 | char* sha256_End(SHA256_CTX*, char[SHA256_DIGEST_STRING_LENGTH]);
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
vendor/trezor-crypto/sha2.c:1231:50: error: argument 2 of type ‘sha2_byte[]’ {aka ‘unsigned char[]’} with mismatched bound [-Werror=array-parameter=]
 1231 | void sha512_Final(SHA512_CTX* context, sha2_byte digest[]) {
      |                                        ~~~~~~~~~~^~~~~~~~
In file included from vendor/trezor-crypto/sha2.c:33:
vendor/trezor-crypto/sha2.h:111:32: note: previously declared as ‘uint8_t[64]’ {aka ‘unsigned char[64]’}
  111 | void sha512_Final(SHA512_CTX*, uint8_t[SHA512_DIGEST_LENGTH]);
      |                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
vendor/trezor-crypto/sha2.c:1250:44: error: argument 2 of type ‘char[]’ with mismatched bound [-Werror=array-parameter=]
 1250 | char *sha512_End(SHA512_CTX* context, char buffer[]) {
      |                                       ~~~~~^~~~~~~~
In file included from vendor/trezor-crypto/sha2.c:33:
vendor/trezor-crypto/sha2.h:112:31: note: previously declared as ‘char[129]’
  112 | char* sha512_End(SHA512_CTX*, char[SHA512_DIGEST_STRING_LENGTH]);
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

This appears to be due to a new warning added to gcc 11.

prusnak, 2021-08-28:
`-Wno-array-parameter` should silence the warning which gets promoted to error via `-Werror`

prusnak, 2021-08-28:
Oddly enough, I don't see any errors when building with `gcc (GCC) 11.1.0`. What version do you use @achow101?

achow101, 2021-08-28:
I am also using 11.1.0

onvej-sl, 2021-09-01:
I can confirm that I have a similar problem with `gcc (Gentoo 11.1.0-r2 p3) 11.1.0`.

onvej-sl, 2021-09-01:
Similar bug occurs also in legacy, see https://github.com/trezor/trezor-firmware/issues/1792.

closed 2021-09-08 by onvej-sl