trezor/trezor-firmware #1788: Problem with compile trezor emulator on linux

TomaszKn, 2021-08-29:
HI
I try compile emulator on my Ubuntu 21.04
I following from this steps:
[https://github.com/trezor/trezor-firmware/blob/master/docs/core/build/emulator.md](https://github.com/trezor/trezor-firmware/blob/master/docs/core/build/emulator.md)
Tan why I run compile (make build_unix) comlication was broken:
`
tomasz@:~/kody/trezor/trezor-firmware/core$ make build_unix
./tools/build_templates
scons -Q -j 4 CFLAGS="-DSCM_REVISION='\"\xeb\x34\xc0\x85\x0e\x8b\xc7\x48\x52\xb5\xf8\xac\xa5\xc3\xab\x78\xdc\x86\x37\x96\"'" build/unix/trezor-emu-core  TREZOR_MODEL="T" BITCOIN_ONLY="0"

python ../common/protob/pb2py /home/tomek/kody/trezor/trezor-firmware/common/protob/messages-binance.proto /home/tomek/kody/trezor/trezor-firmware/common/protob/messages-bitcoin.proto /home/tomek/kody/trezor/trezor-firmware/common/protob/messages-cardano.proto /home/tomek/kody/trezor/trezor-firmware/common/protob/messages-common.proto /home/tomek/kody/trezor/trezor-firmware/common/protob/messages-crypto.proto /home/tomek/kody/trezor/trezor-firmware/common/protob/messages-debug.proto /home/tomek/kody/trezor/trezor-firmware/common/protob/messages-eos.proto /home/tomek/kody/trezor/trezor-firmware/common/protob/messages-ethereum.proto /home/tomek/kody/trezor/trezor-firmware/common/protob/messages-management.proto /home/tomek/kody/trezor/trezor-firmware/common/protob/messages-monero.proto /home/tomek/kody/trezor/trezor-firmware/common/protob/messages-nem.proto /home/tomek/kody/trezor/trezor-firmware/common/protob/messages-ripple.proto /home/tomek/kody/trezor/trezor-firmware/common/protob/messages-stellar.proto /home/tomek/kody/trezor/trezor-firmware/common/protob/messages-tezos.proto /home/tomek/kody/trezor/trezor-firmware/common/protob/messages-webauthn.proto /home/tomek/kody/trezor/trezor-firmware/common/protob/messages.proto --qstr-out build/unix/genhdr/qstrdefs.protobuf.h --bitcoin-only=0

  File "../common/protob/pb2py", line 139
    name: str
        ^

SyntaxError: invalid syntax

python vendor/micropython/py/makeversionhdr.py build/unix/genhdr/mpversion.h

python: can't open file vendor/micropython/py/makeversionhdr.py: [Errno 2] No such file or directory

scons: *** [build/unix/genhdr/qstrdefs.protobuf.h] Error 1

scons: *** [build/unix/genhdr/mpversion.h] Error 2

make: *** [Makefile:152: build_unix] Błąd 2
` `
What I doing wrong?

prusnak, 2021-08-29:
1. Did you clone the repo with recursive flag (to fetch submodules)? `git clone --recursive https://github.com/trezor/trezor-firmware.git`
2. Do you have python3 installed?

closed 2021-08-29 by prusnak

TomaszKn, 2021-08-29:
I clone one again as you wrote and my python version is: Python 3.9.5
Again error:
`  
File "../common/protob/pb2py", line 139`

`    name: str`
`        ^`
`SyntaxError: invalid syntax
`

matejcik, 2021-08-31:
@TomaszKn i believe you're missing the step zero:

> First clone, initialize submodules, install Poetry and enter the Poetry shell as defined here. **Do not forget you need to be in a `poetry shell` environment!**