trezor/trezor-firmware #1796: Substrate (Polkadot) and Sr25519 support

h4x3rotab, 2021-09-04:
I'd love to see Polkadot integration to Trezor.

Unlike most of the blockchains, substrate uses Sr25519 as the signing algorithm by default. Given the C implementation of sr25519 is already [available](https://github.com/TerenceGe/sr25519-donna), it looks possible to port it to Trezor. However if it's not ready, there are still Ed25519 and Ecdsa available.

prusnak, 2021-09-04:
Duplicate of https://github.com/trezor/trezor-firmware/issues/1574

closed 2021-09-04 by prusnak

1000i100, 2022-11-12:
#1574 is closed and locked, i don't find anything about polkadot in trezor-suite, like mentioned in #1574, one more year pass, is there a plan to integrate sr25519 signatures ?

matejcik, 2022-11-14:
There is currently no plan to add DOT support, nor is there any plan for any other use for sr25519.