trezor/trezor-firmware #1800: Some device tests  are working even without enabling passphrase in setup_client

grdddj, 2021-09-07:
**Describe the bug**
In some tests (for example the first testcases in `tests/device_tests/test_protection_levels.py`), it is possible to delete the `passphrase=True` kwarg from the `setup_client` mark, and the tests are still passing. 

**Expected behavior**
Either this kwarg should not be there if it works without it, or it should be failing when we disallow the "passphrase mode".

grdddj, 2021-09-07:
Same happens in https://github.com/trezor/trezor-firmware/blob/f083beb3c26000033c41af2c0f5f1a0c3ae93669/tests/device_tests/test_protect_call.py#L52-L53

matejcik, 2021-09-08:
In case of `test_exponential_backoff_with_reboot`, I indeed see no reason to enable passphrase -- the test is checking that PIN prompt will exponentially increase wait time. This is not touching passphrase at all.

There's apparently a "FIXME TODO add passphrase tests" :)) But I believe that's sufficiently covered elsewhere.

Please modify `test_protect_call.py` as follows:
1. rename to `test_pin`
2. remove the class
3. remove `skip_t2` marker and make sure this runs on TT as well
4. replace `_some_protected_call` with `get_test_address`. No need to check the result, we are interested in the `expected_responses`

wrt (3), there is a difference between T1 and TT: on PIN query, T1 sends `PinMatrixRequest` and TT sends `ButtonRequest(code=PinDialog)`

you can build expected_responses like this:
```
is_t1 = client.features.model == "1"
client.set_expected_responses([
	(is_t1, messages.PinMatrixRequest),
	(not is_t1, messages.ButtonRequest(code=...)),
	messages.Address,
])
```

i'm not sure if TT also returns `PinException` so you'll need to test that

matejcik, 2021-09-08:
in case of `test_protection_levels`, the test is testing that the appropriate protections are executed, so setting passphrase there is right. We need to check that it's there though.

please do the following modifications:
1. remove class
2. add `assert pin_protection is True` and `assert passphrase_protection is True` to every test case where it applies; and `is False` where it doesn't
3. move the `setup_client` marker to top level `pytestmark = ...` so that all testcases are covered unless they have their own different marker
4. try modifying this so that it works on TT as well. instead of the above mentioned trick with `is_t1`, a more appropriate approach here would be something like:
   ```
   def pin_request(client):
	   if client.features.model == "1":
           return PinMatrixRequest
       else:
           ...
   ```
   and then using `pin_request(client)` in the expected_responses.
   (feel free to use this for `test_pin` as well, if you like the solution better)
 
   `test_reset` and `test_recovery` will probably not work on TT, so mark these with `skip_t2` individually
5. replace uses of `btc.get_address` with `get_test_address`

closed 2021-09-17 by matejcik