trezor/trezor-firmware #1808: common ethereum test fixtures

szymonlesisz, 2021-09-14:
Follow up after implementation common fixtures in [trezor-connect](https://github.com/trezor/connect/commit/087c6c2a3497c057abaac1617fd803f8f912b9d3)

- [ ] sign_tx.json (and variations) - consider using hexed values/strings in "value" field. javascript doesnt handle big numbers well (like 12345678901234567890)

- [ ] consolidate sign_tx vs sign_tx_eip1559 result:  `sig_r/sig_s` vs `sig_r_hex/sig_s_hex`

- [ ] consolidate paramaters.path: sometimes there is `m/` (getaddress.json) somtimes there is not (the rest)

- [ ] missing `EthereumGetPublicKey` fixtures

matejcik, 2021-09-14:
> * [ ]  missing `EthereumGetPublicKey` fixtures

we only have one :( if you have more, please contribute

szymonlesisz, 2021-09-14:
we do have 3, on seed `alkohol woman...`
https://github.com/trezor/connect/blob/develop/tests/__fixtures__/ethereumGetPublicKey.js

closed 2021-09-20 by matejcik