trezor/trezor-firmware #1809: AAVE ERC20 Not Supported?

io-raid, 2021-09-15:
Is the Aave erc20 token not supported for anyone else?

I noticed ETHLend (super old Aave) is supported in core/src/apps/ethereum/tokens.py as:
```
elif address == b"\x80\xfb\x78\x4b\x7e\xd6\x67\x30\xe8\xb1\xdb\xd9\x82\x0a\xfd\x29\x93\x1a\xab\x03":
    return TokenInfo("LEND", 18)  # eth / ETHLend
```
but I'm not seeing any support for the official Aave token - 0x7fc66500c84a76ad7e9c93437bfc5ac33e2ddae9

Currently on latest commit on master, 1b932a20e8d235febdb0e3e96e92d4c422ad7f18

Miaalexande, 2021-09-15:
@io-raid You’ll need to reconnect your wallet using the wallet dapps integrator to access the other coins on your wallet

matejcik, 2021-09-15:
Token information is taken from upstream repository https://github.com/ethereum-lists/tokens
It seems that Aave is not listed there; if Aave devs want to be supported on Trezor, they should make sure to add their token information to that repository.

closed 2021-09-15 by matejcik