trezor/trezor-firmware #1816: Fatal error when upgrading firmware: embed/bootloader/messages.o'185

alexgleason, 2021-09-18:
**Describe the bug**
When upgrading my firmware on a Model T from v2.4.0 to v2.4.2 through Trezor Suite (web), it crashed and I saw the following screen:

[image]

I disconnected and reconnected the Trezor, and it was factory reset. I was able to recover my wallet using the mneumonic seed and everything is fine.

**Firmware version and revision**
Upgrading from v2.4.0 to v2.4.2

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Trezor Suite
 - OS: Linux
 - Browser: Brave

prusnak, 2021-09-18:
Thanks for the report. This was caused by unstable USB connection. So this might've been a cable issue or a browser issue. (Brave is not officially supported).

closed 2021-09-18 by prusnak

alexgleason, 2021-09-18:
Brave sucks, so that adds up.