trezor/trezor-firmware #1818: Make use of f-strings in firmware

matejcik, 2021-09-20:
Micropython 1.17 introduces f-string support.

There are a LOT of places in firmware where using f-strings would be practical and make the code more readable. At some point we should mass-convert all format strings to f-string syntax.

mmilata, 2021-09-20:
Just FYI there are a few minor differences from standard/CPython f-strings, see micropython/micropython@692d36d779192f32371f7f9daa845b566f26968d.

amadejpapez, 2021-09-20:
This seems quite straightforward, so I can take this as my first contribution to Trezor if you want. :)

prusnak, 2021-09-24:
> @matejcik This seems quite straightforward, so I can do it as my first contribution to Trezor if you want. :)

Go ahead!