trezor/trezor-firmware #1819: Confirming large Ethereum smart contract requests takes too long

matejcik, 2021-09-20:
In 2.4.2 we introduced universal pagination, meaning that full contents of Ethereum's `data` field is displayed (as hex) to the user.

It turns out that certain smart contracts call a LOT of data, on the order of kilobytes. One Trezor screen fits 90 hex characters, so to confirm 1kB `data` field the user must first swipe through 23 pages.

Possible solutions:

1. when data doesn't fit on screen (or maybe when it's over N pages for N <= 5), ask: "Review 2886 bytes of data? y/n"
2. show first 64 bytes and offer 3 buttons instead of 2: OK, More, Cancel
3. show pagination along with buttons, so you can swipe up/down and also confirm/cancel at any time.

I like option (3) best but i'm not sure how well it can work without the "Swipe" indicator.

matejcik, 2021-09-20:
In general, it would be cool to have a way to exit pagination mid-way through.

For starters, we could add a totally non-discoverable power-user swipe-sideways-to-skip gesture.

gascarcella, 2021-09-21:
As an end-user i would love option 3

karamellar, 2021-09-22:
Yes please, a simple 1inch swap takes forever now 😔

matejcik, 2021-09-22:
Hello dear users, I'm sorry but I'm pre-emptively locking the conversation so that we avoid "me too" spam. Rest assured that fixing this is on the roadmap.

mmilata, 2021-10-20:
For testing:

No data screen shown:
```
trezorctl eth sign-tx -n "m/44'/60'/0'/0/0" --gas-price 1 --gas-limit 2 --nonce 3 0xfc6b5d6af8a13258f7cbd0d39e11b35e01a32f93 123
```
Data fits on confirmation dialog:
```
trezorctl eth sign-tx -n "m/44'/60'/0'/0/0" --gas-price 1 --gas-limit 2 --nonce 3 0xfc6b5d6af8a13258f7cbd0d39e11b35e01a32f93 123 --data 0123456789abcdaa
```
Data does not fit but takes one screen after clicking See All:
```
trezorctl eth sign-tx -n "m/44'/60'/0'/0/0" --gas-price 1 --gas-limit 2 --nonce 3 0xfc6b5d6af8a13258f7cbd0d39e11b35e01a32f93 123 --data 0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa
```
Data takes 4 paginated screens:
```
trezorctl eth sign-tx -n "m/44'/60'/0'/0/0" --gas-price 1 --gas-limit 2 --nonce 3 0xfc6b5d6af8a13258f7cbd0d39e11b35e01a32f93 123 --data 0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa0123456789abcdaa
```