trezor/trezor-firmware #1826: EnvCommandError when running poetry install

iamflorencejay, 2021-09-26:
**Describe the bug**
When I run `poetry install`, this error will be produced every time I want to build from scratch.

```
EnvCommandError

  Command ['/home/user/github/trezor-firmware/.venv/bin/pip', 'install', '--no-deps', 'demjson==2.2.4'] errored with the following return code 1, and output: 
  Collecting demjson==2.2.4
    Using cached demjson-2.2.4.tar.gz (131 kB)
      ERROR: Command errored out with exit status 1:
       command: /home/user/github/trezor-firmware/.venv/bin/python -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-kvbuuap8/demjson_304bc6d5a05e4a259a53c516cc635425/setup.py'"'"'; __file__='"'"'/tmp/pip-install-kvbuuap8/demjson_304bc6d5a05e4a259a53c516cc635425/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-7lwc82nx
           cwd: /tmp/pip-install-kvbuuap8/demjson_304bc6d5a05e4a259a53c516cc635425/
      Complete output (1 lines):
      error in demjson setup command: use_2to3 is invalid.
      ----------------------------------------
  WARNING: Discarding https://files.pythonhosted.org/packages/96/67/6db789e2533158963d4af689f961b644ddd9200615b8ce92d6cad695c65a/demjson-2.2.4.tar.gz#sha256=31de2038a0fdd9c4c11f8bf3b13fe77bc2a128307f965c8d5fb4dc6d6f6beb79 (from https://pypi.org/simple/demjson/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
  ERROR: Could not find a version that satisfies the requirement demjson==2.2.4 (from versions: 1.1, 1.2, 1.3, 1.4, 1.5, 1.6, 2.0, 2.0.1, 2.2, 2.2.1, 2.2.2, 2.2.3, 2.2.4)
  ERROR: No matching distribution found for demjson==2.2.4
  

  at ~/.local/lib/python3.9/site-packages/poetry/utils/env.py:1180 in _run
      1176│                 output = subprocess.check_output(
      1177│                     cmd, stderr=subprocess.STDOUT, **kwargs
      1178│                 )
      1179│         except CalledProcessError as e:
    → 1180│             raise EnvCommandError(e, input=input_)
      1181│ 
      1182│         return decode(output)
      1183│ 
      1184│     def execute(self, bin, *args, **kwargs):
```

**Firmware version and revision**
N/A

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: N/A
 - OS: Manjaro Linux 
 - Browser: Chrome
 - Version: N/A

**To Reproduce**
Steps to reproduce the behavior:
1. Cloning the repo from github
2. run poetry install
3. see error

**Expected behavior**
poetry install should install these deps just fine.

**Screenshots**
N/A

**Additional context**
N/A

prusnak, 2021-09-29:
> * Manjaro Linux

Which version are you at? Is that the latest?

iamflorencejay, 2021-09-29:
> > * Manjaro Linux
> 
> Which version are you at? Is that the latest? If not, please update to latest - 21.1.

I will check on my end.

iamflorencejay, 2021-09-30:
> > * Manjaro Linux
> 
> Which version are you at? Is that the latest? If not, please update to latest - 21.1.

I'm on 21.1.4

gutenye, 2021-10-01:
I have the same issue, looks like [demjson does not support python3](https://github.com/dmeranda/demjson/issues/40), needs to use [demjson3](https://github.com/tmrowco/electricitymap-contrib/pull/3398)

prusnak, 2021-10-01:
Fix in https://github.com/trezor/trezor-firmware/pull/1831

closed 2021-10-01 by prusnak

iamflorencejay, 2021-10-02:
> Fix in #1831

POG