trezor/trezor-firmware #1830: common: stellar test fixtures followup

szymonlesisz, 2021-09-30:
stellar sign tx fixtures from `trezor-common` are now implemented in [connect](https://github.com/trezor/connect/pull/913/commits)

Follow up:
- "timebounds-0-0" case returns different signature in connect for some reason, other tests are working fine.
- `response.signature` in base64 format is required? device/connect returns hex and i see that you are also doing [some conversion](https://github.com/trezor/trezor-firmware/blob/master/tests/device_tests/test_stellar.py#L91)
- [stellar sdk](https://stellar-sdk.readthedocs.io/en/2.11.1/_modules/stellar_sdk/operation/inflation.html) have operation called `inflation` but we dont have `StellarInlnflationOp` is it called differently or not implemented?

matejcik, 2021-10-04:
> * `response.signature` in base64 format is required? device/connect returns hex and i see that you are also doing [some conversion](https://github.com/trezor/trezor-firmware/blob/master/tests/device_tests/test_stellar.py?rgh-link-date=2021-09-30T12%3A11%3A29Z#L91)

base64 is output format of [Stellar Laboratory](https://laboratory.stellar.org/), so the combination of `xdr` and `signature` is easy to verify against the official implementation. I prefer for the computers to do the conversion so that humans don't have to.

> * [stellar sdk](https://stellar-sdk.readthedocs.io/en/2.11.1/_modules/stellar_sdk/operation/inflation.html) have operation called `inflation` but we dont have `StellarInlnflationOp` is it called differently or not implemented?

This is not implemented: https://github.com/trezor/trezor-core/issues/202#issuecomment-393342089
(there's a bunch of other operations that are not implemented, I think)

szymonlesisz, 2021-10-04:
thanks, `timebounds-0-0` with newest emulator returns correct values now,
closing

closed 2021-10-04 by szymonlesisz