trezor/trezor-firmware #1833: Stellar: add support for AMMs related operations.

overcat, 2021-10-06:
**Is your feature request related to a problem? Please describe.**

SDF plans to add support for [AMMs](https://stellar.org/blog/introducing-automated-market-makers-on-stellar) to Stellar Protocol 18, which will be deployed to the test network today. I personally believe that AMMs related features are important for the users and that these operations will be used with high frequency. 

I know you are very busy with Trezor Suit related work right now, **I would love to submit a PR to implement these features**, would you be able to dedicate resources to review it?

If you have any questions, please let me know, thank you.

**Describe the solution you'd like**
- Add support for LiquidityPoolDepositOp and LiquidityPoolWithdrawOp. 
- Extend ChangeTrustOp so that it can add trust lines to liquidity pools.

See: https://github.com/stellar/stellar-protocol/blob/master/core/cap-0038.md#xdr-changes

**Describe alternatives you've considered**
none

**Additional context**

- [Protocol 18 Upgrade Guide](https://stellar.org/developers-blog/protocol-18-upgrade-guide)
- [Introducing Automated Market Makers on Stellar](https://stellar.org/blog/introducing-automated-market-makers-on-stellar)
- [Stellar CAP-0038](https://github.com/stellar/stellar-protocol/blob/master/core/cap-0038.md)

matejcik, 2021-10-06:
For now, we will not be accepting this addition to Stellar. I'm leaving the issue open for now, for if/when the situation changes.

overcat, 2023-09-22:
> For now, we will not be accepting this addition to Stellar. I'm leaving the issue open for now, for if/when the situation changes.

Hi @matejcik, I would like to ask if this status has changed and if you still do not accept the addition of new features.

matejcik, 2023-11-28:
Hello @overcat, the status has indeed changed. Feel free to go ahead with the PR and feel free to open issues for other missing Stellar features.

overcat, 2023-11-29:
Hi @matejcik, thank you for your reply, I have already placed it on my schedule.