trezor/trezor-firmware #1843: Deriving Cardano seed does not show progress bar

matejcik, 2021-10-08:
As of now, Cardano roots are not cached between invocations.

While waiting for the Cardano derivation (that might take several seconds of real time), Trezor screen should show "Please wait..." and a spinner.
Instead, Trezor shows the homescreen until the derivation is done.

tsusanka, 2021-10-12:
I am adding this to the _Code_ column as I believe this is technical.

tsusanka, 2021-11-05:
Will be solved by #1873.

closed 2021-11-10 by matejcik

matejcik, 2021-11-10:
QA:
* open a Cardano wallet (e.g. Adalite)
* a progress bar should be displayed while you're waiting for Trezor to export public keys
* (no need to send any txes so you don't actually need any ADA)

bosomt, 2021-11-16:
@matejcik 

using latest dev master  `trezor-fw-regular-2.4.3-221977ad.bin` i get this error in adalite wallet ;(
do we need to wait till Cardano developers deploy something on their side ?

[image]

`{"exception":{"values":[{"type":"TrezorError","value":"Failed to decode message: Missing required field derivation_type","mechanism":{"handled":true,"type":"generic"}}]},"level":"error","event_id":"f451dbba5dad434584cad0542c6fa597","platform":"javascript","sdk":{"name":"sentry.javascript.browser","packages":[{"name":"npm:@sentry/browser","version":"5.22.3"}],"version":"5.22.3","integrations":["InboundFilters","FunctionToString","TryCatch","Breadcrumbs","GlobalHandlers","LinkedErrors","UserAgent"]},"timestamp":1637076052.7817998,"environment":"production","release":"8.1.2","breadcrumbs":[{"timestamp":1637076032.334,"category":"navigation","data":{"from":"/","to":"/"}},{"timestamp":1637076032.3585,"category":"ui.click","message":"section.welcome > div.welcome-footer > button.button.primary.wide.modal-button"},{"timestamp":1637076034.7577999,"category":"navigation","data":{"from":"/","to":"/"}},{"timestamp":1637076034.7614,"category":"ui.click","message":"section.welcome > div.welcome-footer > button.button.primary.wide.modal-button"},{"timestamp":1637076036.0846999,"category":"ui.click","message":"div.auth-options > div.auth-option.hw-wallet > p.auth-option-text"},{"timestamp":1637076037.8005998,"category":"ui.click","message":"div.trezor-logo-container > svg"},{"timestamp":1637076042.8918998,"category":"fetch","data":{"method":"GET","url":"https://explorer2.adalite.io/api/v2/stakePools","status_code":200},"type":"http"},{"timestamp":1637076049.4972,"category":"ui.click","message":"div.modal-body. > div.modal-content > div.modal-footer > button.button.primary"}],"request":{"url":"https://adalite.io/","headers":{"User-Agent":"Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/95.0.4638.69 Safari/537.36"}}}
`

matejcik, 2021-11-16:
This requires https://github.com/trezor/connect/pull/933 (when that is merged & released, no modifications on adalite side should be necessary)
I thought I mentioned this on the other Cardano issues, maybe I missed this one.

STew790, 2021-12-08:
**QA OK**

Just tested this with @bosomt

PR #1873 (2021-11-10): Cardano derivation improvements
With this PR, the same passphrase is used for Cardano derivation and the standard derivation (fixes #1231). If enabled via `Initialize(derive_cardano=True)`, the Cardano derivation runs at the same time as the standard derivation, which means that the start-up time is longer.

This is only relevant for BIP-39 seeds. With SLIP-39, a different derivation is used and it is available even without `derive_cardano=True`.

Additionally, the Ledger and Icarus derivations are implemented. Ledger derivation is available even in sessions that don't have `derive_cardano=Tru [... cut]