trezor/trezor-firmware #1845: "Forbidden key path" for Avalanche C-Chain derivation path

Mrtenz, 2021-10-10:
**Describe the bug**
Using the Avalanche C-Chain derivation path (`m/44'/9000'/0'/0/0` as per  [SLIP-44](https://github.com/satoshilabs/slips/blob/master/slip-0044.md)) and getting an address from the Trezor device results in a "Forbidden key path" error.

**Firmware version and revision**
Trezor model T on firmware 2.4.2.

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: MyCrypto
 - OS: Arch Linux
 - Browser: Google Chrome
 - Version: 94.0.4606.71

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://app.mycrypto.com/add-account/trezor_new
2. Select Avalanche C-Chain as network
3. Connect to the device

**Expected behavior**
It should be able to fetch addresses.

prusnak, 2021-10-10:
@Mrtenz Ah, I see. You have to submit the coin to the following repo in order to be recognized by Trezor: https://github.com/ethereum-lists/chains (in another word, there is no `eip155-9000.json` file in the repo).

FrederikBolding, 2021-10-10:
@prusnak It does exist, but is missing the `slip44` value: https://github.com/ethereum-lists/chains/blob/master/_data/chains/eip155-43114.json

prusnak, 2021-10-10:
@FrederikBolding Thanks!

@Mrtenz Please update the [chains](https://github.com/ethereum-lists/chains) repo with SLIP44 values you have in your wallet. I thought that Ethereum wallets already do this.

prusnak, 2021-10-10:
@FrederikBolding Thanks!

FrederikBolding, 2021-10-10:
@prusnak It has been merged! So it just needs to be updated on your end! 😄

prusnak, 2021-10-10:
@FrederikBolding Chain/token definitions are updated when the release is being finalized. Next release will happen probably in November or December, so that's when you can expect the definitions will get updated in our repo.

Plus it makes sense to wait with bumping the dependency until @Mrtenz sends his PR to chains repo where they update all SLIP44 numbers they have in MyCrypto.

FrederikBolding, 2021-10-10:
@prusnak I'm a dev at MyCrypto too, just helped out @Mrtenz with this. It's a good idea to review slip44 values, I'll make sure we do that next week 👍

asleson, 2022-02-24:
@prusnak @FrederikBolding 

Hello!  I see that this bug is marked as `Closed` but is it available for general use yet?  I have a Trezor Model T (the newest one) with firmware version `2.4.3`.  However, when using `app.mycrypto.com` and I attempt to add an `AVAX` account (Avalanche C-Chain) from my Trezor, it fails with the "Forbidden key path" error, exactly as described in this bug.  

I'm using `app.mycrypto.com` on Firefox `97.0.1` on Mac OS.

Is the fix not yet available yet?  Or is there another problem?  I'm a software developer by trade myself which is why I have a GitHub account.

Thank you for your help!

FrederikBolding, 2022-02-24:
@asleson Hey there! Sorry about this. There has been some confusion surrounding the Avalanche chains and as such we supported a wrong derivation path. We corrected this but it seems the correction wasn't done on the `chains` repo that Trezor uses. I have opened a PR to rectify this: https://github.com/ethereum-lists/chains/pull/925

asleson, 2022-02-25:
@FrederikBolding Thank you!!  Perhaps you aren't the one to answer this, but that PR that was merged, that's in some shared library right?  So that would need to be pulled in by Trezor in a future firmware release?

FrederikBolding, 2022-02-25:
@asleson You are correct! I believe they update these for each release. But perhaps @prusnak can verify 😄