trezor/trezor-firmware #1878: Device does not react when Simplestaking wallet asks to verify receive address

wendys-cats, 2021-10-29:
**Describe the bug**
When looking at receive address in Simplestaking wallet, wallet informs informs user, after clicking on Receive funds, to interact with the device saying "Password is required. Please wait for Trezor to start.". Device does not react, and user cannot reveal the receive address this way, it has to be manually copied from Wallet page without the verification.

**Firmware version and revision**
Describe firmware version and revision you are using when encountering the bug.

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: SimpleStaking Wallet  (web version)
 - OS: Windows10
 - Browser Chrome Version 95.0.4638.54 (Official Build) (64-bit) + trezord 2.0.27
 - Version [e.g. 90.0.2]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to simplestaking wallet browser version
2. Access accounts from HW 
3. Click on Receive funds
4. Choose the account
5. Wait for "Password is required. Please wait for Trezor to start." message.
6. Wait forever?

**Expected behavior**
Device reacts and shows the receive address.

**Screenshots**
[image]

**Video:** [here](https://watch.screencastify.com/v/IDDe1QkhQUCfPNMdwBBt)

matejcik, 2021-11-01:
Is this a regression? i.e., does the same thing work with older firmware?

my first guess would be a problem in SimpleStaking

wendys-cats, 2021-11-15:
Checked, also present with 2.3.6, 2.3.0, 2.1.6. Should be inform simplestaking?

matejcik, 2021-11-16:
if we have a good way to contact them, yes

i'm closing the firmware issue as this is apparently not a regression and very likely not a problem on our end

closed 2021-11-16 by matejcik