trezor/trezor-firmware #1887: Freezing intermediary FW and its bootloader

hiviah, 2021-11-02:
We discussed freezing of intermediary FW and its bundled bootloader since:

  * We are no longer sending T1 with the old 1.4.0 bootloader
  * For existing units (stored somewhere in drawers) the current functionality to be able to jump upgrade over the bootloaders that had smaller FW size limits is sufficient
  * New FW installed after intermediate FW replaces bootloader that is packed into intermediate FW, upgrading it to latest
  * This would allow to free up space (like old style signatures - `signatures_old_ok`) which are needed only in intermediary FW's bootloader, but would be still needed for rebuild of bootloader for intermediate FW (which would be overflowing preallocated space for bootloader). The old signature style is required there so that seed is not deleted on upgrade from very old FW versions.

matejcik, 2021-11-02:
cc @prusnak for discussion.

ISTM a good solution here is make an `intermediary-fw` branch off the latest intermediary-capable bootloader. If we ever need to update the intermediary firmware, it might be easier to backport the changes to this branch than to keep the necessary functionality in legacy bootloader indefinitely.

In any case, we should establish a sunset timeline for the old signature support (and possibly other legacy features?).

The main problem I can envision is something like, after freeing up the space in bootloader, we develop significant new functionality that we actually need in intermediary, but it won't fit together with the old signature code. That doesn't seem too likely however.

matejcik, 2022-12-08:
This turns out to be impractical.
After changes in #2513, there is no longer the v1 style header recognition. We still do need the v2 style recognition, because _that is what gets installed_ when first upgrading to the v3 bootloader, and the bootloader must be able to recognize the installed image. If we ever change the signature scheme in the future, we _will_ require v3 style recognition, and possibly even v2 style recognition.

Thus, removing the "intermediary" functionality from a production bootloader buys us nothing.

closed 2022-12-08 by matejcik