trezor/trezor-firmware #1888: Show output descriptor on display

andrewkozlik, 2021-11-02:
Trezor should be able to show output descriptors on the display. See https://github.com/bitcoin/bitcoin/blob/master/doc/descriptors.md. We already support output descriptors in trezorlib, see `_get_descriptor()` in `python/src/trezorlib/cli/btc.py`, but the user can only confirm the XPUB on the display, not the entire descriptor.

In `GetPublicKey` replace the `bool` field `ignore_xpub_magic` with an enum field. For example field name `xpub_format` with the values:
0 = `XPUB_VERSION_PREFIX`
1 = `XPUB_PLAIN`
2 = `DESCRIPTOR`
This way the new enum field should be binary-compatible with the current `bool` field. The naming of the field and enum values is open for discussion.

Note that it will probably make sense to do the same for the `ignore_xpub_magic` in `GetAddress`, which is used in multisig.

Hannsek, 2023-07-27:
What it would bring to the end user? What exact information is in the descriptors?

andrewkozlik, 2023-08-29:
> What it would bring to the end user?

It would solve [this](https://github.com/trezor/trezor-firmware/issues/3047#issuecomment-1651677387).

> What exact information is in the descriptors?

Script + XPUB, see https://github.com/bitcoin/bitcoin/blob/master/doc/descriptors.md.

> Can we probably put it only into the qr code?

No. Both places.