trezor/trezor-firmware #1895: Path warning displayed when the same transaction fails later due to strict checks

matejcik, 2021-11-04:
**To Reproduce**

make sure you have strict security: `trezorctl set safety-checks strict`

save the following json:
```
{
  "coin_name": "Testnet",
  "details": {
    "lock_time": 0,
    "version": 2
  },
  "inputs": [
    {
      "address_n": [
        0
      ],
      "amount": 4977040,
      "prev_hash": "a5cd2a706d680587e572df16a8ce5233139a094ebbd148cc66a8004dcc88819c",
      "prev_index": 0,
      "script_type": "SPENDWITNESS",
      "sequence": 4294967293
    }
  ],
  "outputs": [
    {
      "address_n": [
        2147483732,
        2147483648,
        2147483648,
        1,
        0
      ],
      "amount": 4977040,
      "script_type": "PAYTOWITNESS"
    }
  ],
  "prev_txes": {
    "a5cd2a706d680587e572df16a8ce5233139a094ebbd148cc66a8004dcc88819c": {
      "bin_outputs": [
        {
          "amount": 4977040,
          "script_pubkey": "0014820d4a343a44e915c36494995c2899abe3741893"
        },
        {
          "amount": 0,
          "script_pubkey": "6a24aa21a9ed6edfd0a9431aad16549cf318c0d1d237e6a0a87a6403f177caa74fe05546a98f"
        }
      ],
      "inputs": [
        {
          "prev_hash": "0000000000000000000000000000000000000000000000000000000000000000",
          "prev_index": 4294967295,
          "script_sig": "030c15203a205468697320626c6f636b20776173206d696e65642077697468206120636172626f6e206e6567617469766520706f77657220736f75726365201209687a200909200901b34d00641f8d000000",
          "sequence": 4294967295
        }
      ],
      "lock_time": 0,
      "version": 1
    }
  }
}
```

try to sign it: `trezorctl btc sign-tx repro.json`

**Expected behavior:**

When the address `m/0` is encountered, signing should fail immediately with "Forbidden key path"

**Actual behavior:**

First a path confirmation [is displayed](https://github.com/trezor/trezor-firmware/blob/master/core/src/apps/bitcoin/sign_tx/approvers.py#L104-L106). Only afterwards when trying to actually use the keychain, signing fails.

**Ideas to fix:**

Approver (or the `validate_path_against_script_type` helper) should check the value of safety_checks and fail early.

This would disable spending from paths that don't match the script type but are otherwise valid. I'm not sure how big a problem that is.

prusnak, 2021-11-04:
This is already fixed in https://github.com/trezor/trezor-firmware/pull/1891

matejcik, 2021-11-04:
from a cursory read I don't see any modification in the problematic code path so I'm curious as well

prusnak, 2021-11-04:
I thought this diff was responsible, but looking at it again it is responsible for signing messages not transactions:

https://github.com/trezor/trezor-firmware/pull/1891/files#diff-06c957673215089b44cee9cb6687ec24595e6181c6b95625ce1b6e2a88682f82

andrewkozlik, 2021-11-04:
It is fixed in the Taproot implementation thanks to early scriptPubKey derivation. This commit is responsible: 2f1460ed11.

matejcik, 2021-11-04:
oh, nice!

i'll add a testcase after Taproot is merged and then close this issue.

andrewkozlik, 2021-11-04:
> i'll add a testcase after Taproot is merged and then close this issue.

We maybe don't need a new case. I already had to fix the `test_msg_signtx_invalid_path.py-test_invalid_path_fail` UI test due to this: https://github.com/trezor/trezor-firmware/pull/1891/commits/a31cb7c61c6319d39c8cb6f67edc4a9275308465. Maybe just add expected responses.

PR #2017 (2021-12-16): test for: path warning displayed instead of failing immediately
Fixes #1895.

I've chose to `skip_t1` instead of using Litecoin path for the input because T1 does not return error on first `TxAckInput` like TT does.
closed 2021-12-16 by mmilata