trezor/trezor-firmware #1898: ADA Seed Issue

ASBaumgarten, 2021-11-05:
**Describe the bug**
Same seed does not yield the same keys for Cardano between Trezor Model T and the Ledger Nano S. 

**Firmware version and revision**
Firmware version 2.4.2

**Desktop/smartphone setup (please complete the following information):**
 - Wallet software: Adalite (https://adalite.io/)
 - OS: Windows 10 Pro 21H1
 - Browser: Chrome 
 - Version 95.0.4638.69 (Official Build) (64-bit)

**To Reproduce**
Steps to reproduce the behavior:
Use the same 12 word seed phrase to setup both the Model T and the Ledger Nano, and connect devices to Adalite.

**Expected behavior**
The end result of using the same seed phrase should yield identical ADA wallets, but they do not match between the different hardware wallets.

**Additional context**
This issue is somewhat unique to ADA as the same issue is not experienced with BTC or ETH. When viewing the wallet addresses of BTC and ETH on the Model T and Nano S setup with the same seed the same wallet addresses are shown.

ASBaumgarten, 2021-11-05:
Trezor Derivation path: m/1852'/1815'/0'/0/0

Ledger Derivation path: m/1852'/1815'/0'/0/0

prusnak, 2021-11-05:
Yes, Ledger derivations are not compatible with Trezor.

We will add compatibility for Ledger non-standard derivations in one of the upcoming releases - when PR https://github.com/trezor/trezor-firmware/pull/1873 is merged.

Still, the ADA software such as AdaLite would need to specify they want to use Ledger derivation with Trezor.

ASBaumgarten, 2021-11-06:
Greatly appreciate the info. I look forward to the update whenever it's ready for release.

matejcik, 2021-11-06:
duplicate of #1783

closed 2021-11-06 by matejcik