trezor/trezor-firmware #1901: release new production signed bootloader to get touchpad updates

mcudev, 2021-11-07:
I'd like to request that a new signed production bootloader be released in order to get the fix for #1519 into production devices.

It looks like the latest bootloader released with production signatures is 2.0.3 from early 2019 (according to `trezor-firmware/core/embed/bootloader/CHANGELOG.md` and `trezor-firmware/core/CHANGELOG.md`).

That bootloader binary is stored at `trezor-firmware/core/embed/firmware/bootloader.bin` as identified by `trezor-firmware/core/embed/firmware/bootloader_hashes.py` and `trezor-firmware/core/embed/firmware/bl_check.c`, and also by `trezor-firmware/core/tools/headertool.py trezor-firmware/core/embed/firmware/bootloader.bin`.

To note, `trezor-firmware/core/embed/bootloader/version.h` has already been updated to 2.0.4.

A new bootloader will need to be built and signed and then some files, including at least the following, will need to be updated:
- `trezor-firmware/core/embed/firmware/bootloader.bin`
- `trezor-firmware/core/embed/firmware/bl_check.c`
- `trezor-firmware/core/CHANGELOG.md`
- `trezor-firmware/core/embed/bootloader/CHANGELOG.md`

I have an affected device, and I was able to workaround the old bootloader and load a new firmware that contains this fix by opening the case and dropping on a temporary screen + ctpm. After I got the new firmware on the device, then I put the original affected display + ctpm back on and was able to use touch for all the firmware functions, but none of the bootloader functions.

With this new requested bootloader, I could have just grabbed it from the distributed production firmware binary and written it to a microSD, and then loaded the new bootloader onto the device via the microSD + boardloader. That's the way that any other affected users could more easily benefit from the work requested by this issue. Thanks!

hiviah, 2021-11-08:
@prusnak I can do the the necessary steps to get the bootloader included into FW, but need you to sign it. Are you able to sign the bootloader in near future?

So mcudev could take the signed binary from a branch, and we will include the bootloader in next FW release.

prusnak, 2021-11-08:
We need to fix https://github.com/trezor/trezor-firmware/issues/1571 first before releasing the new bootloader. Let's hope this will happen in Q1 2022.

PR #2940 (2023-04-07): include signed bootloader 2.1.0
fixes #1901 

to test it for yourself:

1. go to `SConscript.firmware` line 781 `elif PRODUCTION:` and change the vendorheader to "unsafe_signed_prod"
2. build with `PRODUCTION=1 make build_firmware`
3. run `./tools/headertool.py -D build/firmware/firmware.bin`
4. make upload
closed 2023-04-07 by matejcik