trezor/trezor-firmware #1902: Fix word breaking in TT

andrewkozlik, 2021-11-08:
**Describe the bug**
Word breaking doesn't seem to work correctly for non-ascii text in Trezor T. The dashes are sometimes displayed over the text. Probably a bug in the computation of the text length. See the second and the last line of the message in the screenshot below.

**Firmware version and revision**
trezor-core master branch

**To Reproduce**
Run `test_signmessage_pagination[utf_nospace]` from `test_msg_signmessage.py`.

**Screenshots**
[image]

matejcik, 2021-11-08:
interesting point.
the text rendering function should either properly handle UTF-8, or be marked as accepting `bytes`

matejcik, 2022-12-13:
note to self: investigate whether this problem still exists in UI2

closed 2023-03-16 by Hannsek

andrewkozlik, 2023-03-16:
> It does not appear anymore in UI2.

How do we know?

andrewkozlik, 2023-03-16:
Seriously, you are telling me that in SignMessage we stopped using dashes when breaking long words across lines, and that's the fix? In other words if the bug still exists, then it's just no longer visible in this device test. It does not imply that we are now computing the text length correctly for non-ascii symbols. That sounds to me like we are patching the symptom instead of fixing the cause.

matejcik, 2023-03-17:
did you actually look at the pictures? here's a comparison.
your bug report:
[image]
current rendering:
[image]

notice how the lines in the original report are uneven -- this is partially due to proportional font, but partially, presumably, due to the problem that you are reporting.
in the updated version, the monospace font makes it easier to see that every line renders the same number of **characters**, strongly indicating that the line length computation is correct -- if it weren't, i would expect to see uneven lines again. You can count the number of � by hand to find that the string `PříšerněŽluťoučkýKůňUpělĎábelskéÓdy` also matches the rendering.

Now, I didn't bother actually looking into the rendering code to check if the bug is gone, mostly because I don't know how the bug originally looked like. But I can't reproduce the original problem so :woman_shrugging:

matejcik, 2023-03-17:
or, to put a different tack on the whole thing:

---

**This whole code was thrown out** and replaced with a from-scratch Rust version. The new version does not exhibit the original problematic behavior. Please reopen if you encounter it somewhere else again.

andrewkozlik, 2023-03-17:
> did you actually look at the pictures?

I most certainly did. One can see that in the original the dashes are sometimes displayed over the text as if the last character was not there and sometimes they are displayed perfectly correctly. There is no apparent correlation between the unevenness of the lines and the incorrect placement of the dashes. So "it now aligns nicely" does not say anything about whether the bug is fixed or not. That's first and foremost due to the use of a monospace font in the current rendering. I stand by my comment about "we removed the dashes".

> **This whole code was thrown out** and replaced with a from-scratch Rust version.

That's a fairly legitimate reason to close this issue, as opposed to the reasons stated prior.