trezor/trezor-firmware #1909: Signing message length limit?

bonustrack, 2021-11-11:
Trezor is not triggered when trying to sign a message on snapshot.org, it seem to happen when the length of the message is too long. Is there a length limit of the message that can be signed with Trezor? If so, is it possible to increase that limit?

matejcik, 2021-11-11:
maximum length for `EthereumSignMessage` message is 1024 bytes and it is unlikely to be increased.

please provide more details, maybe steps to reproduce?

volkyeth, 2022-08-11:
> maximum length for `EthereumSignMessage` message is 1024 bytes and it is unlikely to be increased.
@matejcik could you explain why it's unlikely to be increased?

Some platforms are using arbitrarily long signed messages for publishing verified content in the format of posts. Snapshot is one of such.

I'm trying to sign an EIP-712 message, which is 7237 characters long, and Trezor Connect just blinks and closes up without displaying any error, presumably because this is a firmware limitation.

Can that limit be increased?
Is there any workaround for this?

matejcik, 2022-08-11:
EIP-712 is not affected by this limit. Instead, there is a soft limit of ~7 kB per a single text / bytes field. Which is unlikely to be increased either because there is no convincing reason to support 7-kilobyte text fields.

Signing Snapshot.org proposals generally works fine for me, and it seems that Trezor is only signing a hash of the proposal anyway. But you can try to link a proposal that is problematic and I can try to reproduce the problem.

More likely though, the issue is somewhere else. What method do you use to connect Trezor to Snapshot?

volkyeth, 2022-08-13:
My current example for this is actually https://prop.house
Sorry, but you won't be able to reproduce without spamming their platform 😅

They sign the whole content of your proposal. So, the message can be arbitrarily long.

Is there any drawback to allowing the Trezor to sign any messages as long as they fit in memory? I think UX-wise it would have to be a blind signing like for EIP-712 since it would have to paginate through hundreds of pages for long messages

Anyway, I see that this is a narrow edge case, so it's unlikely to be implemented

matejcik, 2022-08-14:
> Is there any drawback to allowing the Trezor to sign any messages as long as they fit in memory?

The "as long as they fit in memory" is the tricky part. Without going into too much detail, it's not as simple as feeding more and more of the message until it doesn't fit.

matejcik, 2022-08-14:
> Is there any drawback to allowing the Trezor to sign any messages as long as they fit in memory?

The "as long as it fits in memory" is the tricky part. Without going into too much detail, it's not as simple as streaming more and more of the message until it doesn't fit anymore. The limits are there to place a safe guarantee that a message that big must work.

With that said, for EIP-712 on the Trezor T,

RealBigPenny, 2022-09-15:
@matejcik as requested I continue here in this closed issue: Other than suggested by your post [here](https://github.com/trezor/trezor-firmware/issues/2516#issuecomment-1247258921) this issue happens with the Trezor Model T and I can replicate it on different devices (3 different computers) with different browser (Firefox, Chormium, Brave), intermediate wallets (Metamask, Rabby, Brave Wallet, Frame) and operating systems (Ubuntu 22.04 / Windows 10). Literally all configurations that do not include a Trezor Model T (I don't have a Trezor One) do not suffer from the 1024 character limit. But if I try a >1024 character message with my Trezor Model T, that exact limit applies.

How can we proceed from here? What information can I provide for you to believe me and so we can debug this issue? Many thanks in advance!

prusnak, 2022-09-15:
@RealBigPenny are you on the latent Trezor firmware release?

RealBigPenny, 2022-09-15:
Yes, sir! @prusnak 
Version 2.5.2

sime, 2022-09-15:
@RealBigPenny it is possible to provide step by step scenario to reproduce this?

Hannsek, 2022-09-16:
It seems that singing ANY message longer than 1024 characters triggers this problem. Testing in progress.

matejcik, 2022-09-16:
fwiw this text does not come from firmware. it might be Connect side, or it might be on Metamask or Snapshot.org side. Perhaps some component proactively handles what it thinks is Trezor's limit.

RealBigPenny, 2022-09-17:
@matejcik it is definitely not the intermediary wallets side. The screenshot I shared is from [Frame](https://frame.sh/) wallet, however the transaction errors out regardless of what you use on the software side. It's also not Snapshot exclusive. Like @Hannsek rightfully says, this error happens for any message longer than 1024 characters that I had tried to sign so far. I just make the Snapshot.org case here because this is severely impacting my capacity to fulfill my duties within my project. Don't you think it's time to open up this issue again?

For what it's worth: I am a huge Trezor proponent due to being an open source maximalist and I am super excited for how Tropic Square will change the hardware landscape for the better! But man, there have been some issues with Trezor in the past and especially this right now is something that leads to people actively being discouraged to buy Trezor devices and opt for Ledger or even worse, use pure software wallets, so they don't get arbitrarily locked out from participating in DeFi and governance. I'm not hear to hate on you guys. I just want to get this issue fixed, so I can confidently push for open source technology within my project and DeFi in general. And of course, to be able to do as I preach and use Trezor to fulfill my duties.

volkyeth, 2022-09-18:
@matejcik 
> fwiw this text does not come from firmware.

Actually, it comes from this line:
https://github.com/trezor/trezor-firmware/blob/0b9f7db742abfc432646f2bd50e0e32b7adaec07/core/src/apps/ethereum/sign_typed_data.py#L427

RealBigPenny, 2022-09-18:
Good find @volkyeth ! :+1: 

And here is the actual hard-coded max limit
https://github.com/trezor/trezor-firmware/blob/0b9f7db742abfc432646f2bd50e0e32b7adaec07/core/src/apps/ethereum/sign_typed_data.py#L36-L37

which disproves yet another claim made by @matejcik : https://github.com/trezor/trezor-firmware/issues/2516#issuecomment-1247258921  

Also pinging @Hannsek since he is actually looking into this issue. :1st_place_medal:

prusnak, 2022-09-18:
> which is 7237 characters long

Meta question: How do you expect an user to go through 7237 characters in order to confirm the transaction? And how do you expect them to do the same on Trezor, where 7237 characters translate to 113 pages on device?

RealBigPenny, 2022-09-18:
Good point, sir. Sunday morning hangover brain. My bad.
The question I have is how 1024 bytes of max message size get treated as 1024 characters when signing messages. [🤔](https://altcodeunicode.com/thinking-face-emoji/)

RealBigPenny, 2022-09-18:
@prusnak Regarding how people are supposed to handle longer text signing transactions: The same way how you present other kinds of long transactions already: 
[image]

volkyeth, 2022-09-18:
Also, EIP-712 is supported by Trezor with blind signing. I guess it's better to be able to sign blindly than not being able to use some dapps.

matejcik, 2022-09-18:
My bad, my bad! Good find @volkyeth 

Well, in that case it's relatively easy to bump that limit to around 7 kB without significant impact, possibly even remove it altogether. @grdddj do you have any memory of why we decided to put this in?

The issue is, of course, that going above said ~7kB maximum will make message signing unreliable. "Some" larger messages will fail to sign for "some" users, depending on the exact structure of the message, size of available RAM, phase of the moon... And the fact that we are actually talking about a 7kB message right here in this thread tells me that sooner or later someone will come back with "and my 37kB proposal fails to sign on Trezor, what a piece of crap".

We could spend some engineering effort to allow for signing arbitrarily long messages, but at that point, if you're pretty much blind signing _anyway_, it seems better to spend the effort on Snapshot.org side and sign a hash of the message instead, which is much more friendly to HW wallets and will also result in significantly faster signing.

Reopening for increasing the max message size a bit.

As for the rest, Trezor was never supposed to work as a blind HSM device for data that the user has no way to actually validate, so adding the "arbitrarily long message" support will have a very low priority.

grdddj, 2022-09-18:
> @grdddj do you have any memory of why we decided to put this in?

Not much, and I could not find any mention of it in the [EIP712 PR](https://github.com/trezor/trezor-firmware/pull/1835)

I think it was kind of a "safety" check to have "some" upper-bound, so `Trezor` will not do something unexpected (or the user will not be "dossed" by a long message)

RealBigPenny, 2022-09-18:
@matejcik just wanted to say thanks for reopening this issue. I took some digs at you because I was annoyed by how you dismissed the message size limitation in past issues. Felt to me very "works for me" attitude, if you know what I mean. I don't say that you actually acted like that, just that I perceived it that way. You opening this issue up again, made me reconsider my initial conclusion. So please accept my apology for being rude.

> "and my 37kB proposal fails to sign on Trezor, what a piece of crap"

I field my issue in part due to me wanting to avoid exactly such comments from others, not least because I have to deal with them myself for being a Trezor proponent. You have every reason to call out Snapshot.org for making that unfortunate design decision. However, since **every other** hardware and software wallet will sign those messages perfectly fine, the only thing people do take away from this is that _"Trezor is broken (again)"_. Especially in a situation like that, it's better to give users the option to **deliberately** compromise, instead of locking them out. That's why I suggested to remove the message size limitation if you disable `safety checks`. Because everybody knows: If you disable safety checks, you are on your f***ing own! So no complaining allowed! :wink: 

Anyway: I hope increasing the limit to 7 kB will help mitigate the issue well enough until a proper size limit bypass has been implemented. Many thanks in advance!

WindyDraft, 2022-10-17:
@matejcik Any update on this issue? I experienced it when trying to update my Snapshot space settings and eventually ended up here after some troubleshooting. I'll also follow up with Snapshot to see if they could do anything on their end.

timjlowe, 2022-11-29:
@matejcik Any update on this?

matejcik, 2022-11-29:
yeah, let's fit this into the next update

tim-becker, 2023-01-05:
Just ran into this, would love to see this fixed

closed 2023-01-13 by grdddj

volkyeth, 2023-01-13:
Is there a tentative release date for this fix?

sime, 2023-01-13:
> Is there a tentative release date for this fix?
> 
> Thanks for seeing this through, btw 🎉

Mid-March is the next planned FW release.

bosomt, 2023-01-25:
**QA OK**

i was able to sign message bigger than 1024B

https://snapshot.org/#/theheaddao.eth/proposal/0x361fc2506c052036a719dffe9dda507b72ebaf2041d3e59cc0970067cf03b0af

 - Device: model T 2.5.4 Universal (revision ab65b18b07170c8a9e4f1506620af988d678453b)