trezor/trezor-firmware #1915: Sign commits and tags

willianpaixao, 2021-11-13:
I'd like to kindly ask developers and maintainers to sign their commits and tags before pushing.

And if I may ask one more thing, enforce that signature in protected branches. This can be done [here](https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/defining-the-mergeability-of-pull-requests/about-protected-branches#require-signed-commits).

willianpaixao, 2021-12-14:
Hi @prusnak @matejcik what are your thoughts on this? It does not require any actual development, just a small fix in the core devs computers. One low-hanging fruit to increase trust.

prusnak, 2021-12-14:
Yes, on our todo list.

matejcik, 2021-12-15:
> Hi @prusnak @matejcik what are your thoughts on this? It does not require any actual development, just a small fix in the core devs computers. One low-hanging fruit to increase trust.

That's very much not true :) Did you know github's Rebase & Merge erases signatures? It would mean either (a) not rebasing anymore and changing the way our git history works, or (b) re-signing every commit after every rebase and then making sure  those can be fast-forwarded onto master. In any case, we are actively evaluating this, but it absolutely does not count as a low-hanging fruit.

willianpaixao, 2021-12-15:
Hi @matejcik, I was unaware of that. And to be honest I don't fully understand what you mean because I do use GPG signature in several repos and never experienced any signature missing. Would you please send me a documentation link on that?

willianpaixao, 2021-12-15:
I think I got what you mean now. When a PR is done, you don't click merge, rather you rebase it to master.

> if you don't use rebases in your project, you won't encounter the problem

I use rebase extensively, no doubt. But the default action for a PR is merge and not rebase for a reason.
Would you care elaborating why this project chooses this?

matejcik, 2021-12-15:
To get that sweet sweet linear history, of course. (Which happens to be much easier to audit.)

prusnak, 2021-12-15:
> To get that sweet sweet linear history, of course.

You can rebase via the CLI which will add your GPG signature to commits. However, I understand that rebase via the GitHub Web UI discards the signatures.

willianpaixao, 2021-12-15:
> To get that sweet sweet linear history, of course

Hum, I see. But in reality, is not really linear, right? It was developed in a different branch, possibly by an outside developer, so it feels like a deviation from the rule (usage of merge) for an esthetic reason.

> Which happens to be much easier to audit
In there I beg to disagree. By rebasing you lose the original branch reference, making traceability harder and therefore audit gets a hit. Not to mention that every other software that is audited would be using rebase if that would help.

But don't get me wrong, this issue is only a suggestion. It's your project. Feel free to close it if you don't feel like agreeing.

matejcik, 2021-12-15:
A significant advantage of linear history is that you can find a specific commit that breaks something. In case of history with merges, that would often be a merge commit, where responsibility can be unclear. By forcing the history to be linear, we are putting the onus on the contributor to make sure their commits are without conflicts, AND that conflicts are resolved at the exact place where they appeared (as opposed to at "end" of the branch). Furthermore, if an interplay of two otherwise non-conflicting changes breaks things, the responsibility can be put on the commit which comes "later" in the linear history -- which is definitely not the case with merges.

In an ideal case this would not affect traceability, as we would ask the contributor to rebase and then fast-forward their commits into master. GitHub unfortunately does not have this functionality built in -- and if it did, it would also not lose signatures from the rebases.

However, auto-signed commits after rebase are a weak form of signing anyway -- unless you manually review every commit after every release, you essentially can't be sure of what you're signing.

I'm not closing the issue, because as I said, we're actively looking into this. I'm just providing context for why it's more than a single-line change in our configs.