trezor/trezor-firmware #1919: Port to Raspberry Pico

willianpaixao, 2021-11-15:
Hi,

as a pet project, I've been porting basic features of the firmware to RP2040, Raspberry Pico's microcontroller.
Mostly the basics, key and mnemonic generation, key derivation, display addresses.

And as I believe this can Raspiberry's developer community into Trezor's codebase, I'd like to ask if those small patches (to the firmware) could be merged to this repository. In another word, if I send PRs would they be accepted?

matejcik, 2021-11-16:
what sort of modifications would that be?

marssystems, 2021-11-16:
Take a look at https://pitrezor.com which is a port to a raspberry pi zero and 4B.

willianpaixao, 2021-11-16:
@matejcik so far has been adding something like

```c
#if defined(RASPBERRYPI_PICO)
#include "pico/stdlib.h"
#endif
```

To later use a function from its library.
Reference [here](https://raspberrypi.github.io/pico-sdk-doxygen/).

@marssystems yes, I know that project for a long time. But this is a little different, since both Pis are running Linux, while Pico is running a bootloader/firmware pretty much like Trezors.

matejcik, 2021-11-23:
we can continue the discussion but i will close this issue for bookkeeping purposes

closed 2021-11-23 by matejcik

prusnak, 2021-11-23:
I guess you want to include `"pico/stdlib.h"` in order to use some functions from that.

Can you indicate how will the changes look when using these functions?

daxime, 2022-04-23:
Hi,
is there any progress in the porting on the Pico ?