trezor/trezor-firmware #1942: ASM Optimizations

FellGleaming, 2021-11-25:
For curiosity sake, I ported the bignum portion of the (older) trezor-crypto repository from 32-bit c++ to 64-bit assembler.  This gave about the performance gains you'd expect from the change.  However, the change to asm allowed me to add further improvements, such as a change from the "nailed" bignum representation to a flat unnailed format; a more efficient squaring algorithm; etc.  Benchmarking the final result showed speed improvements of 6x - 20x, depending on operation.    I don't know if you have considered such assembler optimizations for the current code or not?

invd, 2021-11-27:
I have some questions regarding your efforts:
> (older) trezor-crypto repository

So your changes are against the [trezor-crypto](https://github.com/trezor/trezor-crypto) repository that is two years behind [trezor-crypto](https://github.com/trezor/trezor-firmware/tree/master/crypto) in the monorepo, correct?

> 32-bit c++

`trezor-crypto` is written in C, have you made some other changes to adapt it to C++ first?

> 64-bit assembler

The main use of trezor-crypto is on 32 bit ARM microcontroller platforms (ARM Cortex M3/M4), so a faster x86_64 implementation could not be used there.

I can't speak for the main developers , but there is probably a significant complexity associated with integrating different 64-bit assembler code and hardening it with regards to side channels and testing. I'm not sure if there is sufficient internal or external demand for this.
That said, I am personally interested in such an implementation to reduce runtime overhead in optimized fuzzing builds that trigger a lot of bignum operations, so it would be interesting to see the related code and try out some of your changes.

FellGleaming, 2021-11-27:
Yes, the changes are against the older repo, in X86_64 asm.  I didn't mean to suggest that this particular code could be used in monorep; it was done merely to see what performance increases one could expect from optimizations such as an unnailed bignum format, etc.    It was a quick-and-dirty port for benchmarking only.

However, unless the interfaces have changed dramatically, the "bignum" abstraction is well-encapsulated from the rest of the codebase, so that porting just this portion to 32-bit ARM assembler is very feasible, and, I estimate, the algorithmic improvements this would allow would double performance, possibly more.  I wanted to gauge any possible interest in adding a conditional-build option that used these optimizations.

I'll contact you directly regarding the existing x86_64 code.

prusnak, 2021-11-27:
What invd says above. 

We won't be needing x86_64 ASM optimizations in our codebase because our code is meant to be run on 32-bit ARM only.

Moreover. we recently switched to [secp256k1 library](https://github.com/bitcoin-core/secp256k1) for secp256k1 operations.

I am closing this issue. That said, please send any interesting code to invd and have a look at secp256k1 library above. Maybe you will see ways how to optimize that and that would be great since everybody could benefit from that change.

closed 2021-11-27 by prusnak