trezor/trezor-firmware #1943: Implement message signing for Taproot

andrewkozlik, 2021-11-25:
First we need to decide how to deal with Taproot addresses in `SignMessage` and `VerifyMessage`, what to hash and whether to tweak the keys. Bitcoin Core supports message signing only for legacy addresses. Other wallets either don't support Taproot at all or they don't implement message signing for Taproot addresses. We will wait with the implementation in Trezor until there is some consensus in the community about the way this should work.

Some resources mention using BIP-322, but the specification still isn't finished and the last time we used BIP-322 for ownership proofs in SLIP-19, the whole BIP-322 specification got rewritten.
https://bitcoinops.org/en/topics/generic-signmessage/
https://bitcoinops.org/en/newsletters/2021/09/29/#preparing-for-taproot-15-signmessage-protocol-still-needed

This is also related to AOPP: https://github.com/trezor/trezor-firmware/issues/1586.

mmilata, 2022-12-05:
There seems to be active effort getting BIP-322 support into Bitcoin Core: bitcoin/bitcoin#24058.

AlexanderPavlenko, 2024-05-05:
Will implementing this also add support for Nostr events sign/verify? Seems like a nice to have feature.
NIP-01: "Signatures, public key, and encodings are done according to the [Schnorr signatures standard for the
 curve `secp256k1`](https://bips.xyz/340)".

```shell
# test mnemonic from NIP-06: leader monkey parrot ring guide accident before fence cannon height naive bean
trezorctl device recover
trezorctl set safety-checks prompt
trezorctl btc sign-message -t taproot -n "m/44'/1237'/0'/0/0" '[0,"17162c921dc4d2518f9a101db33695df1afb56ab82f5ff3e5da6eec3ca5cd917",1703260028,1,[],"hello world"]'
```

Expected result close to `"sig": "9c3ba7ca1b4942f784c014b28779c4a904c902c75104fc0ae72485d0dfac27ce95e1178cc33eb86d4d86ec354004a77e965553dfafb01f47733a6b31d79cc56c"`, got `Error: ProcessError: Unsupported script type` instead.

psychoet, 2024-06-03:
No more update on this?

Hannsek, 2024-06-03:
Not yet…

bkiac, 2025-01-28:
@Hannsek some wallets, like Sparrow, support simple BIP-322 signatures for single-key spend Taproot addresses. Is this out of scope for Trezor? I know BIP-322 hasn’t gained much traction yet, but BIP-137 also took time and still isn’t in Bitcoin Core (and likely won’t be).

What would need to happen for Trezor to add BIP-322 signing to the firmware?

prusnak, 2025-01-29:
> What would need to happen for Trezor to add BIP-322 signing to the firmware?

Let's say 3 major software wallets support this. Then we can consider adding the support.

andrewkozlik, 2025-01-29:
>Will implementing this also add support for Nostr events sign/verify?

Exposing the raw BIP340 singing functionality without enforcing domain separation sounds like a dangerous step to me. Which is what you'd need to sign arbitrary stuff, such as Nostr events. Unfortunately it seems the authors of NIP-01 didn't make any effort to enforce domain separation, which doesn't help. If we want to support Nostr signing, we should have a whole separate app for that in Trezor anyway.

As for BIP-322 I have to say I am still not at all happy with the spec.
- It is still a draft.
- There is only one test vector.
- Maybe I need to read it more carefully, but it seems to me that when signing Proof of Funds the value of `message_challenge` is not well defined.
- I am not sure whether the rule for using the simple signature type for P2WSH-P2SH is correctly defined. I would have to give that more thought.

All this indicates to me that it is still a work in progress. That being said, it's the closest thing to a standard for P2TR message signing that we have.

guggero, 2026-09-08:
> As for BIP-322 I have to say I am still not at all happy with the spec.

I agree, that's why I pushed for the spec to be finalized: https://github.com/bitcoin/bips/pull/2141

Which changes all your bullet points from above:

* It is still a draft. **--> In state "Complete" now**
* There is only one test vector. **--> Extensive test vectors for all three sub formats now exist, see https://github.com/bitcoin/bips/tree/master/bip-0322**
* Maybe I need to read it more carefully, but it seems to me that when signing Proof of Funds the value of `message_challenge` is not well defined. **--> This has been addressed specifically as well. PSBTs for BIP322 signing requests now have a new global `PSBT_GLOBAL_GENERIC_SIGNED_MESSAGE` field that carries the message.**

So the only missing part is @prusnak's requirement: "Let's say 3 major software wallets support this. Then we can consider adding the support."
I'm working on that one. Maybe with the spec being in a better shape now, we don't need to wait for 3 wallets?
The implementation risk IMO is smaller if the spec isn't in a draft state that can change at any time.