trezor/trezor-firmware #1947: Investigate how to save flash space

prusnak, 2021-11-26:
We are slowly running out of Flash space. Let's collect some ideas how to save it:

- [ ] compress embedded bootloader (see below)

prusnak, 2021-11-26:
We can save some space by compressing embedded bootloader (using the same method as we do for TOIF):

``` python
>>> import zlib
>>> d = open("bootloader.bin","rb").read()
>>> z = zlib.compressobj(level=9, wbits=-10)
>>> c = z.compress(d) + z.flush()
>>> len(d)
94208
>>> len(c)
61366
```

This will save us around 33KB of Flash space, but increases the complexity of critical part of the firmware - updating the bootloader.

hiviah, 2021-12-06:
I experimented with `secp256k1_ecmult_static_context` since it seems it would be a good candidate given its size and would fit CCMRAM exactly when uncompressed/generated on start. It turns out it doesn't compress well, but generating does not seem that slow, estimated ~85 ms. It's also hassle, but wanted to explore another option.

grdddj, 2022-03-31:
Running `nm` on the current `master` emulator found out the `Rust UI` itself has also around `27 kb`:
```
...
536: _ZN10trezor_lib2ui9component4text10paragraphs19Paragraphs$LT$T$GT$13change_offset17had74e3db161430f3E
541: _ZN109_$LT$trezor_lib..ui..component..base..Child$LT$T$GT$$u20$as$u20$trezor_lib..ui..layout..obj..ObjComponent$GT$9obj_event17hbee63d60a5accfe6E
567: _ZN112_$LT$trezor_lib..ui..model_tt..component..swipe..Swipe$u20$as$u20$trezor_lib..ui..component..base..Component$GT$5event17hdcc1f376b17c79f6E
615: _ZN109_$LT$trezor_lib..ui..component..base..Child$LT$T$GT$$u20$as$u20$trezor_lib..ui..layout..obj..ObjComponent$GT$9obj_place17h1cfdb8671da4a2f6E
789: _ZN109_$LT$trezor_lib..ui..component..base..Child$LT$T$GT$$u20$as$u20$trezor_lib..ui..layout..obj..ObjComponent$GT$9obj_event17hee566170cee1a9a7E
904: _ZN109_$LT$trezor_lib..ui..component..base..Child$LT$T$GT$$u20$as$u20$trezor_lib..ui..layout..obj..ObjComponent$GT$9obj_paint17h4de8d47d3d7eca79E
941: _ZN10trezor_lib2ui9component4text6layout10TextLayout11layout_text17he0b38fe77ad4ed96E
950: _ZN109_$LT$trezor_lib..ui..component..base..Child$LT$T$GT$$u20$as$u20$trezor_lib..ui..layout..obj..ObjComponent$GT$9obj_event17hb733cb801aba63d4E
1154: _ZN109_$LT$trezor_lib..ui..component..base..Child$LT$T$GT$$u20$as$u20$trezor_lib..ui..layout..obj..ObjComponent$GT$9obj_event17hfe386ae64e4c874fE
1234: _ZN10trezor_lib2ui9component4text9formatted26FormattedText$LT$F$C$T$GT$14layout_content17he18e49b6355d36caE
1526: _ZN109_$LT$trezor_lib..ui..component..base..Child$LT$T$GT$$u20$as$u20$trezor_lib..ui..layout..obj..ObjComponent$GT$9obj_event17h580e53b1b2d37313E
27500: total
167: total_amount
```

grdddj, 2022-04-01:
I continued with analyzing `build/firmware/frozen_mpy.o`, which has information about the size of `micropython` functions, and it is possible to get the size of each application in `scr/apps` (and also each function in those applications).

Helper script for analysis is at https://gist.github.com/grdddj/5aeb38f17713f038639dde1c667dbd21

Notable results and examples:
```sh
>>> python get_apps_size_from_binary.py --statistics
...
11_876: apps/stellar
11_987: trezor/crypto
14_849: apps/nem
20_187: apps/management
27_881: apps/webauthn
29_693: apps/common
35_966: apps/cardano
59_706: apps/bitcoin
59_985: trezor/ui
63_002: apps/monero
101_749: apps/ethereum
516_077: total
```

```sh
>>> python get_apps_size_from_binary.py --lines
...
1092: core/src/apps/webauthn/fido2.py
1098: core/src/apps/monero/signing/step_09_sign_input.py:28  ... sign_input (1)
1175: core/src/apps/monero/xmr/mlsag.py:342 ... _generate_clsag
1371: core/src/apps/webauthn/knownapps.py:20  ... by_rp_id_hash
1379: core/src/trezor/ui/layouts/tt/__init__.py
1392: mp_frozen_mpy_content
1493: core/src/trezor/enums/MessageType.py
1691: core/src/apps/cardano/sign_tx.py
2378: core/src/apps/monero/xmr/bulletproof.py:1509 ... BulletProofBuilder.verify_batch (1) (1)
7176: core/src/apps/ethereum/networks.py:44  ... _networks_iterator
7561: core/src/all_modules.py
8092: core/src/apps/ethereum/tokens.py:14  ... token_by_chain_address - CONST_TABLE_DATA (SPECIAL)
8361: core/src/apps/common/coininfo.py:92  ... by_name
10241: mp_frozen_mpy_names
27184: mp_qstr_frozen_const_pool
37077: core/src/apps/ethereum/tokens.py:14  ... token_by_chain_address
555386: total
13369: total_amount
```

```sh
>>> python get_apps_size_from_binary.py --app ethereum --lines
...
304: core/src/apps/ethereum/sign_tx_eip1559.py
305: core/src/apps/ethereum/sign_tx.py
368: core/src/apps/ethereum/sign_typed_data.py:443 ... validate_field_type
406: core/src/apps/ethereum/layout.py
427: core/src/apps/ethereum/sign_typed_data.py
428: core/src/apps/ethereum/networks.py:44  ... _networks_iterator - CONST_TABLE_DATA (SPECIAL)
451: core/src/apps/ethereum/sign_tx_eip1559.py:55  ... sign_tx_eip1559
455: core/src/apps/ethereum/sign_tx.py:35  ... sign_tx
650: core/src/apps/ethereum/sign_typed_data.py:227 ... TypedDataEnvelope.get_and_encode_data
7176: core/src/apps/ethereum/networks.py:44  ... _networks_iterator
8092: core/src/apps/ethereum/tokens.py:14  ... token_by_chain_address - CONST_TABLE_DATA (SPECIAL)
37077: core/src/apps/ethereum/tokens.py:14  ... token_by_chain_address
101749: total
2493: total_amount
```

```sh
>>> python get_apps_size_from_binary.py --file-name ethereum/sign_typed_data.py --lines
...
81:  core/src/apps/ethereum/sign_typed_data.py:392 ... write_leftpad32
119: core/src/apps/ethereum/sign_typed_data.py:204 ... TypedDataEnvelope.find_typed_dependencies
121: core/src/apps/ethereum/sign_typed_data.py:183 ... TypedDataEnvelope.encode_type (1)
125: core/src/apps/ethereum/sign_typed_data.py:41  ... sign_typed_data
126: core/src/apps/ethereum/sign_typed_data.py:516 ... get_name_and_version_for_domain
148: core/src/apps/ethereum/sign_typed_data.py:141 ... TypedDataEnvelope._collect_types
173: core/src/apps/ethereum/sign_typed_data.py:350 ... encode_field
230: core/src/apps/ethereum/sign_typed_data.py:414 ... validate_value
234: core/src/apps/ethereum/sign_typed_data.py:61  ... generate_typed_data_hash
368: core/src/apps/ethereum/sign_typed_data.py:443 ... validate_field_type
427: core/src/apps/ethereum/sign_typed_data.py
650: core/src/apps/ethereum/sign_typed_data.py:227 ... TypedDataEnvelope.get_and_encode_data
4428: total
94:  total_amount
```

```sh
>>> python get_apps_size_from_binary.py --grep trezor_ui --lines
...
415: core/src/trezor/ui/components/tt/scroll.py:352 ... paginate_paragraphs
428: core/src/trezor/ui/layouts/tt/reset.py:26  ... show_share_words
431: core/src/trezor/ui/components/tt/passphrase.py
447: core/src/trezor/ui/layouts/tt/__init__.py:794 ... confirm_properties (1)
458: core/src/trezor/ui/layouts/tt/__init__.py:617 ... confirm_blob (1) (1)
533: core/src/trezor/ui/components/tt/scroll.py
537: core/src/trezor/ui/__init__.py
657: core/src/trezor/ui/style.py
662: core/src/trezor/ui/components/common/text.py:180 ... render_text
809: core/src/trezor/ui/components/tt/reset.py:42  ... Slip39NumInput.on_render
1379: core/src/trezor/ui/layouts/tt/__init__.py
59985: total
1556: total_amount
```

There could be a lot of conclusions from this, I would note the size of `trezor/ui`, almost `60 kb`, which could compensate for the `Rust UI`

mmilata, 2022-04-12:
We can save 12KiB by using different implementation of blake2b. The reference implementation we use does some [manual inlining](https://github.com/trezor/trezor-firmware/blob/core/v2.4.3/crypto/blake2b.c#L201) that can be turned into loop, slowing the function down somewhat. [This one](https://www.oryx-embedded.com/doc/blake2b_8c_source.html) doesn't use such inlining.

matejcik, 2022-05-17:
it appears that a lot of debug dump functionality from Rust is getting into the builds. we might want to keep an eye on that.

things that seemed to help locally:
- instead of `#[derive(Debug)]`, use `#[cfg_attr(debug_assertions, derive(Debug))]` ([source](https://users.rust-lang.org/t/will-derive-debug-be-compiled-in-release-build/30388)
  - this will break (some) bare `unwrap()`s because `Result<T, E>::unwrap()` requires `E: Debug`
- replace bare unwraps with something like `unwrap_or_panic(error: &'static str)`

more generally, we need to watch for the presence of these things in the binaries. `nm` can see them. 

---

Another thing to work on is some sort of dependency tracker for included symbols. This is more important for the bootloader usecase, where just plain including Rust costs us over 10 kB of space. Using `nm` together with `objdump` disassembly would allow us to construct a dependency graph with weights, identify functions that are only used by a subtree, etc.

---

One more thing to watch out for is generic instance explosion -- code that is succinct in Rust might end up generating a lot of needless repetitions in the binary.

A thing we should investigate: whenever const generics are used (such as `Field<32>`), we _might_ end up with multiple instances for `Field<1>.read()`, `Field<20>.read()` etc. I don't know if the compiler is smart about this, we might end up extracting the const generic to a regular struct member.

mmilata, 2022-05-19:
> A thing we should investigate: whenever const generics are used (such as Field<32>), we might end up with multiple instances for Field<1>.read(), Field<20>.read() etc. I don't know if the compiler is smart about this, we might end up extracting the const generic to a regular struct member.

Investigated a bit and it appears that all generic parameters get monomorphized, including consts. [Assembly output seems to confirm this](https://godbolt.org/z/3qE8n6Ks5). So I'd consider de-constifying the parameter.

matejcik, 2022-12-13:
We are now doing reasonably well on flash space. This issue is not actionable and once we dump all relevant info into Notion we will close it.

closed 2023-01-18 by matejcik