trezor/trezor-firmware #1966: Do not show "go to trezor.io/start" notification

matejcik, 2021-12-02:
When the device is initialized, the bootloader should not trigger the Chrome pop-up that directs users to trezor.io/start.

This may be already done. We need to investigate the current status and implement the feature if it's not there.

Might be done in concert with #1049

tsusanka, 2021-12-02:
I think this was done by https://github.com/trezor/trezor-firmware/commit/a4ba24645b65b870aaf55e79d0e52c7be57b2519? Which is from March 4th 2021 and it seems [latest BL release was March 2019](https://github.com/trezor/trezor-firmware/blob/91dd21b561b7f9a0a0d34fc694bd739e80dbe6f1/core/embed/bootloader/CHANGELOG.md#203-march-2019)?

prusnak, 2021-12-02:
> I think this was done by [a4ba246](https://github.com/trezor/trezor-firmware/commit/a4ba24645b65b870aaf55e79d0e52c7be57b2519)?

This is something different.

The relevant change (for core) happened in https://github.com/trezor/trezor-firmware/commit/07cd73ce57e4e6842ceb5eb5b0162d71f27bc8e2

prusnak, 2021-12-02:
The relevant change (for legacy) happened in https://github.com/trezor/trezor-firmware/commit/215020a592a4ee7e51c64a29e7f6cfeaf21a2a3d (legacy 1.9.0 and later, that is the bootloader 1.8.0 does not have this feature, the bootloader 1.10.0 does have this feature).

matejcik, 2021-12-02:
Thanks for the info. Closing this as a duplicate of #841.

closed 2021-12-02 by matejcik