trezor/trezor-firmware #1974: Metamask freezes when trying to confirm transaction on Trezor T

Fefiox, 2021-12-04:
When trying to confirm the transaction after giving access to metamask to my trezor T wallet, it freezes and doesn´t continue, freezing also the browser.

Steps to reproduce (REQUIRED)

    Make a transaction
    Opens a new page where metamask asks for permission to confirm the transaction on the Trezor T
    Give access
    When the page closes and the transaction should appear to confirm on the Trezor, nothing happens and the app freezes.

Expected behavior
Confirm the transaction on the Trezor T.

Browser details (please complete the following information):

    OS: Windows
    Hardware Wallet: Trezor Firmware version 2.4.2
    Browser: Firefox
    MetaMask Version:10.6.4

prusnak, 2021-12-04:
Already tracked here: https://github.com/MetaMask/metamask-extension/issues/12967

closed 2021-12-04 by prusnak

Einoow, 2021-12-09:
Same problem, waiting for fix.