trezor/trezor-firmware #1983: Protobuf message is interpreted twice

tsusanka, 2021-12-06:
**Describe the bug**
Firing `RebootToBootloader` followed by `GetFeatures` instantly lead to a peculiar behaviour where the user dialog (Do you want to restart...) is shown again (and `ButtonRequest` is sent again as well).

**Firmware version and revision**
I have encountered this with 1.10.1 but I think this will concern all >1.10.0 devices.

**Desktop/smartphone setup (please complete the following information):**
- trezorlib on current master (4419572437f7f3f4239a03725cc1902f5803fc1e)
- macOS m1

**To Reproduce**
```
from trezorlib.client import get_default_client
from trezorlib.log import enable_debug_output
from trezorlib import messages
from time import sleep

enable_debug_output()
client = get_default_client()
client.call(messages.RebootToBootloader())

client.call(messages.GetFeatures())
```

Output log can be found [here](https://gist.github.com/tsusanka/59fe58a68b9d13bcea980a579de7053f).

**Expected behavior**
Firmware responds with `Features` on `GetFeatures` message.

**Additional context**
- Note that I believe this issue can only be replicated on fast computers (whatever that means) when the second message is fired instantly. 
- I am not sure if this is bound to the `RebootToBootloader` message, maybe this is a generic problem.
- We solve this issue in Connect by some sleep so it is not a time critical issue, however if it can be solved, it would be quite nice as the results are unexpected.

hiviah, 2021-12-06:
Yes the sleep is there to flush response message out (but there's no "flush and USB is now stopped". On first look the usbSleep should only wait. Will investigate further why it seems to recurse handler.

hiviah, 2021-12-06:
The issue is caused by combination of two things:

We can't do

  1. receive message
  2. reset
  3. send response

So 2. and 3. are swapped.

Using usbSleep really does process messages, but incorrectly, as we broke the FSM by calling `dwc_poll` while still inside FSM.

Dialog shown first time:

```
[0] from 0x0801a684 in fsm_msgRebootToBootloader+0 at fsm.c:351
[1] from 0x08018e56 in msg_process+66 at messages.c:229
[2] from 0x08018f36 in msg_read_common+182 at messages.c:283
[3] from 0x0804ac6a in dwc_poll+470 at ../../usb/usb_dwc_common.c:401
[4] from 0x08018b50 in usbPoll+12 at usb.c:410
[5] from 0x0801a176 in main+166 at trezor.c:185
```

Second time recursed from `fsm_msgRebootToBootloader`:

```
[0] from 0x0801a684 in fsm_msgRebootToBootloader+0 at fsm.c:351
[1] from 0x08018e56 in msg_process+66 at messages.c:229
[2] from 0x08018f36 in msg_read_common+182 at messages.c:283
[3] from 0x0804ac6a in dwc_poll+470 at ../../usb/usb_dwc_common.c:401
[4] from 0x08018bba in usbSleep+30 at usb.c:456
[5] from 0x0801a6de in fsm_msgRebootToBootloader+90 at fsm.c:364
```

Using USB-less sleep will cause the response message not to be sent (and client will throw LIBUSB error).

If we set some global flag `reset_from_main` and do reset in `main`, it would not show the dialog twice, but it would still eat up any following messages until finally reset comes. I will look if we can somehow make the USB flush. However, the correct communication from client should wait let's say a second-two after receiving reset success message, otherwise sending quickly messages after reset will always result in race condition - you never know when you hit the state between reset and device ready.

prusnak, 2021-12-06:
+1 for "solving" this by always waiting 2 seconds after `RebootToBootloader` is sent.

hiviah, 2021-12-06:
I fixed the double dialog appearance for `RebootToBootloader`. First version reorders read and write by USB handler calling reset if a flag is set from FSM (#1984).

Second version is cleaner, just writes out all pending packets before reset (#1985).

In any case, the client still needs to wait after receiving `Success` on `RebootToBootloader`, otherwise it'll get `LIBUSB_ERROR_IO`.

tsusanka, 2021-12-07:
Ok great, this is basically how we solved it in Connect, see

https://github.com/trezor/connect/blob/c454341f9ef6805cdf9d0b57ba24eacef7ffb56f/src/js/core/Core.js#L658-L667

So I will change it to one full second (@hiviah or more? It seems like 0.5s did do the trick as well) and alter the comment with a link to this issue as it incorrectly blames trezord. Sounds good?

hynek-jina, 2021-12-09:
@tsusanka Please provide instruction for QA

hiviah, 2021-12-09:
@tsusanka 
> So I will change it to one full second (@hiviah or more? It seems like 0.5s did do the trick as well) and alter the comment with a link to this issue as it incorrectly blames trezord. Sounds good?

501 ms is minimum, 1 s should be OK. Although waiting 501 ms, then doing let's say 5 retry calls to GetFeatures until you get response instead of libusb error spaced 100 ms between each retry would make it slightly faster.

tsusanka, 2021-12-11:
Changed to 1s in https://github.com/trezor/connect/pull/972.

tsusanka, 2021-12-11:
QA:
- Please make sure firmware upgrades works on 1.10.* firmwares in Suite and the device is correctly rebooted to bootloader automatically.

tsusanka, 2021-12-13:
So as https://github.com/trezor/trezor-firmware/pull/1985 is merged, I believe this is done. QA will test after.

closed 2021-12-13 by hynek-jina