trezor/trezor-firmware #1987: messages_pb2.py

BEHNAMKK1, 2021-12-07:
hi 
when i build.docker.sh  and then run poetry run ./script/cibuild
i have this error
  CC      ../layout.small.o
  CC      ../memory.small.o
  CC      ../oled.small.o
  CC      ../random_delays.small.o
  CC      ../rng.small.o
  CC      ../setup.small.o
  CC      ../supervise.small.o
  CC      ../timer.small.o
  CC      ../usb21_standard.small.o
  CC      ../usb_standard.small.o
  CC      ../util.small.o
  CC      ../webusb.small.o
  CC      ../winusb.small.o
  CC      ../vendor/trezor-crypto/bignum.small.o
  CC      ../vendor/trezor-crypto/ecdsa.small.o
  CC      ../vendor/trezor-crypto/hmac_drbg.small.o
  CC      ../vendor/trezor-crypto/memzero.small.o
  CC      ../vendor/trezor-crypto/secp256k1.small.o
  CC      ../vendor/trezor-crypto/sha2.small.o
  LD      bootloader.elf
  OBJCOPY bootloader.bin
./firmware_align.py bootloader.bin
make: Leaving directory '/media/behnam/01D7EB443F8C6CF0/ubuntu/aaa_2/trezor-firmware/legacy/bootloader'
make: Entering directory '/media/behnam/01D7EB443F8C6CF0/ubuntu/aaa_2/trezor-firmware/legacy/firmware/protob'
  PROTOC  messages_pb2.py
messages.proto:215:14: Missing numeric value for enum constant.
Makefile:40: recipe for target 'messages_pb2.py' failed
make: *** [messages_pb2.py] Error 1
make: Leaving directory '/media/behnam/01D7EB443F8C6CF0/ubuntu/aaa_2/trezor-firmware/legacy/firmware/protob'

matejcik, 2021-12-07:
your `protoc` version is too old. (see e.g. https://github.com/wechaty/grpc/issues/116)

closed 2021-12-07 by matejcik

taherfattahi, 2023-05-03:
Install `protobuf-compiler` on your OS:
`sudo apt-get remove protobuf-compiler`