trezor/trezor-firmware #1997: Stellar transaction sending fails with current public model T firmware

bosomt, 2021-12-08:
**Describe the bug**
Stellar transaction sending fails with current public firmware.

**Firmware version and revision**
2.4.3 public version revision 0x595b14254c1abb2be3f69e42c7932f1eca8cf1b1

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'https://accountviewer.stellar.org/dashboard'
2. Sign/send stellar transaction 
3. See error

**Screenshots**
[image]

bosomt, 2021-12-09:
**QA NOK**

Tested in Exodus with same firmware version 2.4.3.
Signing/sending end in error.

[image]

cryptomar1o, 2021-12-10:
Downgrading FW version to 2.4.2 does not solve the issue with Stellar Viewer, but it is now possible to sign a transaction with Exodus

❌ 2.4.3 + Stellar viewer
❌ 2.4.3 + Exodus
❌ 2.4.2 + Stellar viewer
✅ 2.4.2 + Exodus

puruoni, 2021-12-10:
Trezor one is also the same, when I send it after signing with trezor-connect, I get the response "tx_bad_auth".

I only checked the firmware versions 1.10.3 and 1.10.4, but I got the same error. Too sudden, but is this caused by stellar network? Or trezor?

matejcik, 2021-12-10:
From what I can tell, the Account Viewer problem lies on Stellar side.

I am able to build and successfully send a virtually identical transaction with the same account on Testnet. However, Mainnet won't accept it. For now, I have no reason to believe that the signature is actually invalid.

---

Exodus appears to not send the `asset` field. We originally didn't require it, and assumed `native` if not specified. We now require it to be specified. This will need to be fixed on Exodus side. Do we have any way to contact them?

prusnak, 2021-12-10:
> From what I can tell, the Account Viewer problem lies on Stellar side.

FTR - Account Viewer repo is here: https://github.com/stellar/account-viewer-v2

matejcik, 2021-12-10:
> FTR - Account Viewer repo is here: https://github.com/stellar/account-viewer-v2

yeah but this isn't an Account Viewer problem either, looks more like their backend is hating on them or something.

SpartanKang50, 2021-12-10:
With my Trezor and Stellarterm not communicating properly, my crypto is inaccessible and I’m losing thousands … what can be done about this?  Is there any other way to access my funds??

felixlucid, 2021-12-11:
> With my Trezor and Stellarterm not communicating properly, my crypto is inaccessible and I’m losing thousands … what can be done about this? Is there any other way to access my funds??

Downgrading to Trezor Model T 2.4.2 does not solve the issue with StellarTerm. Assets held on SDEX remain entirely inaccessible via Model T on devices at any time upgraded to 2.4.3

prusnak, 2021-12-11:
It seems this will be fixed when we fix the issue in Trezor Connect - https://github.com/trezor/connect/issues/970

matejcik, 2021-12-11:
closing in favor of trezor/connect#970
wrt the Exodus issue, I've contacted them so hopefully they can fix this on their side.

closed 2021-12-11 by matejcik