trezor/trezor-firmware #1998: EIP-1559 support

cambuehl, 2021-12-08:
Trezor model t does not support eip-1559, this is a bug that needs to be corrected immediately.  Without it the trezor products are completely irrelevant.

prusnak, 2021-12-08:
Trezor Model T supports EIP-1559 since firmware 2.4.2 released last month.

closed 2021-12-08 by prusnak

cambuehl, 2021-12-08:
Then why am I constantly getting this message:  Invalid transaction params: params specify an EIP-1559 transaction but the current network does not support EIP-1559?  Also, why is a trezor dev by the name of matejcik, on your forums telling people: “EIP-1559 for Trezor One will be included in next firmware update, which is currently scheduled for December 8th – that is, if everything goes right.”  Obviously trezor 2.4.2 never supported eip-1559, and neither does the newest update.  Your own developers have admitted this.  You should be ashamed of yourself for closing this issue, it seems all you’re trying to do is make it difficult for new customers to see the problems that are associated with your devices.  This is a major issue and I’m not the only one having this problem.  It needs to be fixed!  Downgrading metamask should not be a workaround you’re telling people to use, you need to upgrade your products!

cambuehl, 2021-12-08:
Then why am I constantly getting this message:  Invalid transaction params: params specify an EIP-1559 transaction but the current network does not support EIP-1559?  Also, why is a trezor dev by the name of matejcik, on your forums telling people: “EIP-1559 for Trezor One will be included in next firmware update, which is currently scheduled for December 8th – that is, if everything goes right.”  Obviously trezor 2.4.2 never supported eip-1559, and neither does the newest update.  Your own developers have admitted this.  You should be ashamed of yourself for closing this issue, it seems all you’re trying to do is make it difficult for new customers to see the problems that are associated with your devices.  This is a major issue and I’m not the only one having this problem.  It needs to be fixed!  Downgrading metamask should not be a workaround you’re telling people to use, you need to upgrade your products!

Closed #1998<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Ftrezor%2Ftrezor-firmware%2Fissues%2F1998&data=04%7C01%7C%7C3670ea31ed3a432fe65408d9ba7a3f9b%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637745856480526283%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=o0EbF4dbP%2F7B2210hK4Tkb9%2FEVJkASdlGRnfjkdoP7Q%3D&reserved=0>.