trezor/trezor-firmware #1999: EIP-1559 support is needed

cambuehl, 2021-12-08:
There is still no support for eip-1559 after the latest firmware update.  This is an update that is already 5 months late.  As each day passes, trezor products become more and more useless and irrelevant because you simply refuse to bring your products up to date.  You people need to bring your products up to date, this is the biggest update to the eth network, and it was know for months in advance.  This update should've been ready for your products the day after eip-1559 was implemented, you had months of notice, there's no excuse.  Yet here we are 5 months after and still no update.  When do you expect you'll actually get this update done?  It's completely unacceptable, and to be frank, it's making trezor look very bad!

prusnak, 2021-12-08:
Trezor Model T supports EIP-1559 since firmware 2.4.2 released last month.
Trezor Model One supports EIP-1559 since firmware 1.10.4 released today.

closed 2021-12-08 by prusnak

cambuehl, 2021-12-08:
Then why am I constantly getting this message:  Invalid transaction params: params specify an EIP-1559 transaction but the current network does not support EIP-1559?  Also, why is a trezor dev by the name of matejcik, on your forums telling people: “EIP-1559 for Trezor One will be included in next firmware update, which is currently scheduled for December 8th – that is, if everything goes right.”  Obviously trezor 2.4.2 never supported eip-1559, and neither does the newest update.  Your own developers have admitted this.  You should be ashamed of yourself for closing this issue, it seems all you’re trying to do is make it difficult for new customers to see the problems that are associated with your devices.  This is a major issue and I’m not the only one having this problem.  It needs to be fixed!  Downgrading metamask should not be a workaround you’re telling people to use, you need to upgrade your products!