trezor/trezor-firmware #2001: EIP-1559 transactions don't wotk with trezor, but fine with ledger

cambuehl, 2021-12-08:
If you've truly implemented support for eip-1559, then why am I constantly getting this message:  Invalid transaction params: params specify an EIP-1559 transaction but the current network does not support EIP-1559?  Also, why is a trezor dev by the name of matejcik, on your forums telling people: “EIP-1559 for Trezor One will be included in next firmware update, which is currently scheduled for December 8th – that is, if everything goes right.”  Obviously trezor 2.4.2 never supported eip-1559, and neither does the newest update.  Your own developers have admitted this.  You should be ashamed of yourself for closing this issue, it seems all you’re trying to do is make it difficult for new customers to see the problems that are associated with your devices.  This is a major issue and I’m not the only one having this problem.  It needs to be fixed!  Downgrading metamask should not be a workaround you’re telling people to use, you need to upgrade your products!  All of these transactions that fail with trezor, work flawlessly with my ledger.  Why aren't you able to complete the same transactions as ledger wallets?  It seems your devs are getting there messgaes crossed.

AJolly, 2021-12-08:
I'm not part of the team, but I've seen metamask fail with trezor/eip-1559 tx's, but frame.sh work fine, try that instead?

prusnak, 2021-12-08:
You are mixing Trezor One and Trezor Model T updates. On firmware side, everything is ready. For Metamask you need to wait for the next Metamask update: https://github.com/MetaMask/metamask-extension/issues/12130

closed 2021-12-08 by prusnak