trezor/trezor-firmware #2008: poetry run ./script/cibuild

BEHNAMKK1, 2021-12-10:
when i run poetry run ./script/cibuild 
i get this error

  AS      header.o
  CC      messages.o
  MAKO    coin_info.h
  CC      config.o
  CC      trezor.o
  CC      pinmatrix.o
  MAKO    nem_mosaics.h
  CC      fsm.o
In file included from fsm.c:373:0:
fsm_msg_common.h: In function 'fsm_msgResetDevice':
fsm_msg_common.h:305:3: error: implicit declaration of function 'reset_init' [-Werror=implicit-function-declaration]
   reset_init(msg->has_display_random && msg->display_random,
   ^~~~~~~~~~
fsm_msg_common.h: In function 'fsm_msgEntropyAck':
fsm_msg_common.h:317:3: error: implicit declaration of function 'reset_entropy' [-Werror=implicit-function-declaration]
   reset_entropy(msg->entropy.bytes, msg->entropy.size);
   ^~~~~~~~~~~~~
fsm_msg_common.h: In function 'fsm_msgBackupDevice':
fsm_msg_common.h:329:5: error: implicit declaration of function 'reset_backup' [-Werror=implicit-function-declaration]
     reset_backup(true, mnemonic);
     ^~~~~~~~~~~~
cc1: all warnings being treated as errors
../Makefile.include:224: recipe for target 'fsm.o' failed
make: *** [fsm.o] Error 1
make: Leaving directory '/media/behnam/01D7EB443F8C6CF0/ubuntu/aaa_2/trezor-firmware/legacy/firmware'

prusnak, 2021-12-10:
This is not a support portal. Please join our community discussion on Gitter to get help with building our project: https://gitter.im/trezor/community

prusnak, 2021-12-10:
Also duplicate of https://github.com/trezor/trezor-firmware/issues/1995

closed 2021-12-10 by prusnak