trezor/trezor-firmware #2009: support for EOSIO chains

dafuga, 2021-12-10:
Hey guys, not sure if this is the best place to ask this, but I was wondering if anyone here could give some idea of the level of effort that would be required to add support for an EOSIO chain - https://eos.io/

Also, would some of the core devs be interested in maybe taking on such a task? If so, how much would a project like that cost us?

prusnak, 2021-12-10:
EOS is already supported, see https://trezor.io/coins/#EOS

closed 2021-12-10 by prusnak

dafuga, 2021-12-11:
Thanks for your reply. I did see that document earlier, but it seems to indicate that EOS isn't supported on the Trezor One. Is that something that you guys are currently working on adding?

prusnak, 2021-12-11:
That is not planned. Use Trezor Model T if you need to interact with EOS.

dafuga, 2021-12-13:
And would you guys be opened to adding support for the Trezor Model One if we were going to fund its development? I'm just trying to get an understanding of our options.

prusnak, 2021-12-13:
> for the Trezor Model One

We are currently not interested in adding coins that are not in the TOP 30 on Coinmarketcap. Should this situation change in the future we might reconsider.