trezor/trezor-firmware #2011: Broken links in changelogs

prusnak, 2021-12-11:
Github stopped redirecting from e.g. https://github.com/trezor/trezor-firmware/issues/1227 to https://github.com/trezor/trezor-firmware/pulls/1227 and vice-versa :-(

We kind of relied on it in our changelogs as we always used the `/issues/` in our changelog URLs.

What needs to be fixed now:
 
- [ ] fix links in `core/CHANGELOG.md`
- [ ] fix links in `core/embed/boardloader/CHANGELOG.md`
- [ ] fix links in `core/embed/bootloader/CHANGELOG.md`
- [ ] fix links in `core/embed/bootloader_ci/CHANGELOG.md`
- [ ] fix links in `legacy/bootloader/CHANGELOG.md`
- [ ] fix links in `legacy/firmware/CHANGELOG.md`
- [ ] fix links in `legacy/intermediate_fw/CHANGELOG.md`
- [ ] fix links in `python/CHANGELOG.md`
- [ ] update `tools/generate-changelog.py` to generate `/pulls/` in the URL instead of `/issues/` when the referenced number is PR and not an issue

Carlosbogo, 2021-12-11:
Hi, I would like to work on this. Is there anyone doing it already?

prusnak, 2021-12-11:
@Carlosbogo Not at this moment, please, go ahead!

commit dd1240ef (2021-12-17): docs: update links in changelogs (#2013)

fixes #2011
closed 2021-12-17 by prusnak