trezor/trezor-firmware #2020: EIP-1559 transactions on TT do not require hold 5 seconds to confirm

yura-pakhuchiy, 2021-12-15:
Transaction is confirmed immediately after user confirms totals fees. This is different from how confirmations works for legacy ETH transactions and for other coins and may lead to confirmation of unwanted transaction, due to user may expect one more screen for final confirmations, or due to poor quality touchscreen in TT (mine sometimes registers click in wrong location or double clicks instead of single click).

bosomt, 2021-12-25:
**QA NOK**

tested by 
`trezorctl eth sign-tx -n "m/44'/60'/0'/0/0" -e 2 --nonce 1234567890 --gas-limit 123 --max-gas-fee 234 --max-priority-fee 456 0x27b1fdb04752bbc536007a920d24acb045561c26 4000`

yes there ara only confirmation screens but not hold to confirm screen
https://photos.app.goo.gl/BQrTtSYqzyqpd3St7

device 2.4.3 fw public revison 0x595b14254c1abb2be3f69e42c7932f1eca8cf1b1

closed 2022-02-08 by matejcik

bosomt, 2022-02-14:
**QA OK**

hold to confirm displayed during
` trezorctl eth sign-tx -n "m/44'/60'/0'/0/0" -e 2 --nonce 1234567890 --gas-limit 123 --max-gas-fee 234 --max-priority-fee 456 0x27b1fdb04752bbc536007a920d24acb045561c26 4000`

FW version : trezor-fw-regular-2.4.4-d979efc3.bin 0xd979efc3cac49ce3c3ee99ce4dbba6e6217844d8

PR #2103 (2022-02-08): fix(core/ethereum): implement hold-to-confirm for eip1559 transactions
fixes #2020 

[image]