trezor/trezor-firmware #2028: memory protect =1

BEHNAMKK1, 2021-12-18:
i program board with memory protect=1 
can i reset board and program board again?

prusnak, 2021-12-18:
No, that is not possible. MEMORY_PROTECT=1 locks the board and disables the JTAG completely.

closed 2021-12-18 by prusnak

BEHNAMKK1, 2021-12-18:
when i program board i have this error on lcd 
broken firmware detect
unplug your trezor ,reinstall firmware 
how can i fix this problem?

prusnak, 2021-12-18:
You have uploaded the custom bootloader which is not the official one. You need to compile the firmware with MEMORY_PROTECT=0 to disable the bootloader check and use that custom-built firmware.

BEHNAMKK1, 2021-12-18:
i am program board with MEMORY_PROTECT=1
can program firmware board with MEMORY_PROTECT=0?

prusnak, 2021-12-18:
> can program firmware board with MEMORY_PROTECT=0?

Yes, compile firmware with MEMORY_PROTECT=0 and upload it via regular firmware update (`trezorctl firmware-update`)

BEHNAMKK1, 2021-12-19:
when run 
trezorctl firmware-update -f
i have this error 
Traceback (most recent call last):
  File "/usr/bin/trezorctl", line 32, in <module>
    from trezorlib.client import TrezorClient, TrezorClientDebug, CallException
ImportError: cannot import name 'TrezorClientDebug'

BEHNAMKK1, 2021-12-19:
after trezor to bootloader mode and run this line 
trezorctl firmware-update -f /media/behnam/B48A28E88A28A8B6/wallet/1_1/trezor-firmware-memory-protection==0/legacy/firmware/trezor.bin
i get this error
Trezor One v2 firmware (1.8.0 or later)
Firmware version 1.10.5 build 0
No signatures found. Continue? [y/N]: y
Invalid firmware data.
Firmware validation failed, aborting.