trezor/trezor-firmware #2031: Support ZIP-244 TxIDs

andrewkozlik, 2021-12-20:
The NU5 network upgrade in Zcash introduces a new TxID digest [1]. The legacy TxIDs will still be accepted by the network, but in order for Trezor to be able to spend UTXOs from transactions which use the new ZIP-244 digest, Trezor will apparently need to implement the new algorithm, so that it may verify previous transaction output amounts. The NU5 upgrade should activate on Zcash Mainnet on 18th April 2022 [2].

- In trezor-core the `Zcashlike` class will need to override `get_prevtx_output()` to compute the legacy TxID or ZIP-244 TxID depending on the transaction version. 
- The same will be needed in trezor-legacy.
- Updates may also be needed in trezor-suite and blockbook. Further investigation is needed.

[1] ZIP-244: https://zips.z.cash/zip-0244#txid-digest
[2] https://z.cash/support/schedule/

jarys, 2021-12-22:
I'm working on ZIP-244 support for TT. I will submit it for review in the first half of January.

jarys, 2022-01-03:
ZIP-244 will be modified in order simplify the signing process on HWW. They will include `sha_amounts` and `sha_scriptpubkeys` fields introduced BIP-341. [[source](https://github.com/zcash/zips/issues/574#issuecomment-1002803827)]

prusnak, 2022-02-23:
@hynek-jina Why is this marked as `blocked`?

@martinboehm Are there any changes on the blockbook side of things? I don't remember exactly if we had to do any changes in blockbook related to previous network upgrades. Can you please deploy Zcash Testnet blockbook to see if it works as intended?

hynek-jina, 2022-02-23:
@prusnak 
I thought it was blocked by the activation on the Zcash Mainnet planned on April.

prusnak, 2022-02-23:
> I thought it was blocked by the activation on the Zcash Mainnet planned on April.

We need to implement this before the Zcash upgrade is in effect not after.

martinboehm, 2022-02-23:
@prusnak the testnet is running (behind firewall) here https://blockbook-dev.corp.sldev.cz:19132/ and there are version 5 transactions already, for example https://blockbook-dev.corp.sldev.cz:19132/tx/68ac26aebc57e1a0bfa89dc4f52c816d92c3606de7209e0bb4c6286e4739c1c3.